rkaczorek/astroberry-server #1: possible update issues

abordercollie, 2017-05-17:
This was on a fresh image. 
There was an issue in the update. Rerunning allowed it to continue

astroberry@astroberry:~$ sudo astroberry-update
[sudo] password for astroberry: 
**/usr/local/bin/astroberry-update: line 81: unexpected EOF while looking for matching `"'
/usr/local/bin/astroberry-update: line 84: syntax error: unexpected end of file**
astroberry@astroberry:~$ sudo astroberry-update
Updating Astroberry Server...
Astroberry Server updated successfully

There seems to be an issue in the upgrade script:

astroberry@astroberry:~$ sudo astroberry-upgrade
Your system version is 1.0.1 and the latest version is 1.0.3
Do you want to upgrade your system (y/n)? y
Upgrading to version 1.0.2...
Updating PHD2...
Fixing hostname...
Updating noVNC...
Updating custom packages...
Installing Planetary Imager...
Your system has been upgraded to version 1.0.2
Upgrading to version 1.0.3...
Updating Planetary Imager...
**/usr/local/bin/astroberry-upgrade: line 159: apt-get-y: command not found**
Installing Skychart/Cartes du Ciel...
Installing INDI Web Manager...
Created symlink from /etc/systemd/system/multi-user.target.wants/indiwebmanager.service to /etc/systemd/system/indiwebmanager.service.
Updating noNVC...
Synchronizing state of nginx.service with SysV init with /lib/systemd/systemd-sysv-install...
Executing /lib/systemd/systemd-sysv-install enable nginx
Setting up welcome page...
Your system has been upgraded to version 1.0.3
From now on visit http://astroberry.local to access welcome page and https://astroberry.local/desktop/ to access Astroberry Server
Your system is up to date.

rkaczorek, 2017-05-17:
You can ignore errors in astroberry-update.
For the astroberry-upgrade the issue has been just fixed. A typo in the upgrade script resulted in not installing libqhy. Please install libqhy manually by running: sudo apt-get -y libqhy

closed 2017-05-17 by rkaczorek

abordercollie, 2017-05-17:
Thank you. Will do.

This is the line that worked: sudo apt-get -y install libqhy

rkaczorek, 2017-05-17:
This is right.