rkaczorek/astroberry-server #2: Error on login

HamHamFonFon, 2018-03-01:
Hi,

First, great job for your application :).

When i want to log in via web browser on  https://<my-IP-local>/desktop, i have this on login screen : 
`noVNC encountered an error:
Uncaught TypeError: Cannot read property 'appendChild' of null
https://astroberry.local/desktop/app/custom_ui.js:834:40
TypeError: Cannot read property 'appendChild' of null at Object.setAlienStyles (https://astroberry.local/desktop/app/custom_ui.js:834:40) at Object.initSettings (https://astroberry.local/desktop/app/custom_ui.js:192:16) at Object.start (https://astroberry.local/desktop/app/custom_ui.js:114:16) at Object.WebUtil.initSettings (https://astroberry.local/desktop/app/webutil.js:165:22) at Object.load (https://astroberry.local/desktop/app/custom_ui.js:104:21) at window.onscriptsload (https://astroberry.local/desktop/app/custom_ui.js:75:45) at HTMLScriptElement.loadFunc (https://astroberry.local/desktop/app/webutil.js:409:28)`

Few weeks ago, i have only change wifi configuration to allow my rPi to connect to my local network and sometime i update packets via synaptics.
If you have some questions, i will trying to answer you.

Thx

rkaczorek, 2018-03-01:
I'm aware of this issue. It is already fixed in the development version. And it will be released along version 1.0.5 soon. As a workaround just refresh your browser at the login screen a few times until the error does not pop up. Then you can login without any problem.

HamHamFonFon, 2018-03-02:
Okay thank you for your answer. I'm waiting for the next upgrade :).

th3boz, 2018-07-10:
Hi!
When I try to connect via web to https://astroberry.local I see the homepage but when I enter the password (astroberry) it says a generic “Failed to connect to server”.
Is there a log where I can see what’s wrong? Thanks!!

th3boz, 2018-07-10:
Well.. an additional (and I hope uselful) information: if I try to connect to home wifi (via vlanconf-gui) I don’t see any wifi network.. and so the only way to connect to astroberry is via Ethernet or via VAP. Thanks!

rkaczorek, 2018-07-10:
Hi th3boz
The first issue seems to be related to x11vnc server is not starting at boot. You can check this by running: sudo systemctl status x11vnc.service
If it's dead just restart it by running: sudo systemctl restart x11vnc.service
I'm working on the newest version and have just noticed that there is some display-manager dependence. If so edits to /etc/systemd/system/x11vnc.service would be needed.
I'm aware of the second issue and I'm planning to fix it in the next release. You can still use wlanconf in your terminal to connect to your home wlan.

th3boz, 2018-07-10:
Hi rkaczorek,
thank you for the so quickly reply.
Unfortunately the x11vnc service is up’n’running :(
Well.. I try to restore the original SD image provided by you because I did a lot of update (about 500) doing both apt-get upgrade and apt-get dist-upgrade.. and I don’t knowledge now it was a good idea ;)
I’ll give you a feedback when I restored to the original state.
Thank you.

th3boz, 2018-07-11:
Well.. with the original image version I have the same problem.
For info.. I try to connect to the webinterface via iPad (on witch I cannot import the certificate): I see the homepage but I have an error on login.
I tried to use also Chome, IE and Edge on Windows 10, but I cannot see the homepage for an SSL Error (even if I import the certificate in the trusted certification authority).

rkaczorek, 2018-07-11:
This is coming from the way browsers handle self-signed and local certificates. There's no known workaround except not using ssl at all... Use http://astroberry.local:8080/index.html url to access astroberry server without ssl. This method disables side panel for GPS and INDI Webmanager, but it's not a big deal. I will address it in the next release. It should be available soon. I'm waiting for official release of Ubuntu Mate 18.04 so we can use the latest LTS version in astroberry.

th3boz, 2018-07-12:
Thanks very much.. it works on 8080. I solved the SSL replacing the self-signed certificate with my certificate (that is not self-signed) and now it works.
I also suggest to you another improvement: a bigger /boot partition (now when I do via GUI the updates it says that there is only 42MB free and is not enought).
Thanks again for you job!!

rkaczorek, 2018-07-12:
Great! The size of the boot partition is a known issue of Ubuntu Mate, which is a base for Astroberry Server. Thanks for your input, much appreciated.

FreemanZlat, 2018-08-01:
Hi rkaczorek,
I try to connect http://astroberry.local from my laptop using Chrome browser, but it shows an ERR_CONNECTION_REFUSED error. http://192.168.10.1 shows homepage, but https://192.168.10.1/desktop/ shows SSL error (even after installing the certificate).
Ping to astroberry.local (it's resolved as 127.0.1.1) works fine.

jochym, 2018-08-01:
You cannot use the certs/ssl properly on local addresses. There is no chain of trust for names to build secure connection the way https is designed. If you need security on internal networks use http over secure tunnels like openvpn or ssh.

rkaczorek, 2018-08-02:
FreemanZlat, please use a workaround described in my comment above. As I said before ssl will be disabled in the next release due to reasons also pointed out by jochym.

rkaczorek, 2018-11-04:
Fixed in version 1.1.0

closed 2018-11-04 by rkaczorek