rkaczorek/astroberry-server #4: Image file larger than 16 SD Card

th3boz, 2018-07-08:
Hi rkaczorek
thank you very much for your fantastic job!! I'd like to use the astroberry system for controlling my scope and I think that it's really wonderflul!!
I downloaded the image (and checked the checksum), but when I try to write it to my 16GB SD Card (with Win32DiskImager) it says that the image is a little bit larger than the SD Card; a 16GB SD card has 14.4 free space while the image is 14.8GB.
I tried to write it ignoring the warning, but when I start Pi there is immediately a kernel panic "Kernel Panic: VFS : Unable to mount root fs on unknown-block (179,2)".
Could you please help me with this issue?
Thanks very very much!!!

rkaczorek, 2018-07-09:
Well, it happens (rarely) in case of some SD cards... You can shrink astroberry server image before writing it to SD card. Take a look at this example https://softwarebakery.com//shrinking-images-on-linux

th3boz, 2018-07-10:
Great! Thank you very much! I’m going right now to try the system! Great job!!!

rkaczorek, 2018-11-04:
Fixed in version 1.1.0

closed 2018-11-04 by rkaczorek