rkaczorek/astroberry-server #8: System boot partition too small

peterstellar, 2018-09-11:
Hi Rkaczorek,

This project looks great, thank you for all your hard work.
I have done the install and updated the app but when I run Software Updater for the system I get "Not enough free disk space" error message. The current image has 63Mb for the Boot Sector but needs about 67Mb for the latest software setup. My resizing of the boot sector with Gparted has failed several times. Is there a simple solution to this?
[image]

Thanks Peter

peterstellar, 2018-09-13:
While the issue is not with Astroberry Server, as soon as someone downloads Astroberry, installs it and does a system update they're likely to encounter this problem, using the current image.
I tried the SD card partition resize again (on my notebook) using Gparted with the same result.
There are a couple more screenshots attached if they help, along with the partition check file
[image]
[image]
[Gparted Details.txt](https://github.com/rkaczorek/astroberry-server/files/2378904/Gparted.Details.txt)
If you think this problem may be a wider issue, would you please consider increasing the size of the boot partition on your image so that it is less likely to happen again. If it is just me, then I'll select the updates I think I'll need and go with that.
Thanks for your prompt reply and consideration.

rkaczorek, 2018-09-15:
I cannot figure out why the error occurs while resizing the partition. I would recommend you partition a SD card manually - just create boot partition and root partition setting whatever size you want, name the partitions, create file systems on the partitions (fat16 on boot, ext2 on root), set the partition flags, copy the files from original Astroberry Server partitions. This will remove the issue.
The next release of Astroberry Server will address this.

rkaczorek, 2018-11-04:
Fixed in version 1.1.0

closed 2018-11-04 by rkaczorek

peterstellar, 2018-11-05:
Thank you for the update and all your help. I downloaded the image
this morning and have it running well. Regards Peter

----- Original Message -----
From: "rkaczorek/astroberry-server" 
To:"rkaczorek/astroberry-server" 
Cc:"peterstellar" , "Author" 
Sent:Sun, 04 Nov 2018 13:02:56 -0800
Subject:Re: [rkaczorek/astroberry-server] System boot partition too
small (#8)

	Closed #8 [1]. 

	—
thread [3]. 

Links:
------
[1] https://github.com/rkaczorek/astroberry-server/issues/8
[2]
https://github.com/rkaczorek/astroberry-server/issues/8#event-1945014331
[3]