rkaczorek/astroberry-server #13: Low Voltage Warning leads to instability issues

MartinOST, 2019-02-13:
Fresh install of Astroberry image on Raspberry Pi3. During power on stage, a low voltage warning indicator is being printed on the screen. Once the Desktop environment is ready, the yellow voltage indicator on the right appears. 

This (https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=58151) might suggest the image is using old boot code but I cannot verify this. 

Currently, my Pi is rebooting approx. every 5 mins. I can increase this frequency by inducing load on the system (i.e. running KStars).

rkaczorek, 2019-02-13:
What power supply do you use? Have you mounted radiators on CPU and GPU?

MartinOST, 2019-02-13:
I am using a canakit power supply from this kit:

https://www.amazon.com/CanaKit-Raspberry-Complete-Starter-Kit/dp/B01C6Q2GSY

After the instability issues began, I unplugged all peripheral devices except mouse and keyboard. 

The psu indicates 5V 2.5a but I do not have a meter to verify this.

rkaczorek, 2019-02-14:
2.5A should be enough. I can't reproduce this issue on neither 3B nor 3B+.
The boot is new so 3B+ can be supported. What I would recommend is to check RPi's temperature with
`/opt/vc/bin/vcgencmd measure_temp`

The correlation of load and instability looks like there's a problem with CPU temperature on your board.
Also run stock raspbian OS for verification.

MartinOST, 2019-02-14:
Thanks for the lead. 

I did put the small passive coolers on both chips as recommended by Canakit. 

I could also verify that Chrome seems to be the almost instant death of the Pi. I have run the stock Raspian for ~3hrs before installing Astroberry. 

I'll measure the temps tonight and will report back!

rkaczorek, 2019-02-14:
What happened after these 3 hrs with raspbian?

MartinOST, 2019-02-14:
It was a new Pi I had purchased just for the purpose of installing AstroBerry. After playing with Raspian for a few hours, I decided to install AstroBerry. while playing with Raspian, I noticed no instability whatsoever.

rkaczorek, 2019-02-14:
Thanks for the feedback. Let's check the temperature first. We'll try to find the solution.
I'm using it on two of my scopes both on 3B and 3B+ and have had no problems with stability at all.

MartinOST, 2019-02-14:
Yeah, if this would be a more wide-spread issue, you'd see people complain a lot more. I think my problem is an isolated one. I will report back as soon as I can read the temps.

MartinOST, 2019-02-15:
I did some test runs. 

Idle temp is ~43C

Carte duCiel brings it up to ~48, same with Kstars. 

Chrome brings it up to ~56C. However, pretty much consistently, Chrome will cause the Pi to reboot. Either upon starting of Chrome or when I navigate to a different web page. 

I'd use the built in firefox but that crashes upon startup as well.

rkaczorek, 2019-02-15:
It's a bad news then. The temperatures are by far normal.

Crashing firefox unfortunately is also normal - the one originally distributed with Ubuntu Mate has this problem, you can replace it manually with the one from Ubuntu:
`wget http://ports.ubuntu.com/pool/main/f/firefox/firefox_52.0.2+build1-0ubuntu0.12.04.1_armhf.deb
sudo apt-get -y purge firefox
sudo dpkg -i firefox_52.0.2+build1-0ubuntu0.12.04.1_armhf.deb
sudo apt-mark hold firefox`

The only thing that comes to my mind is to try other SD card. If it doesn't help it must be some kind of a hardware problem. On the other hand it does not appear on Raspbian... Very strange :-/

MartinOST, 2019-02-15:
To add to the confusion ... 

I just took the SD card and plugged it into another Pi that I had ordered. Brand new as well. 

Boots up just fine _without_ low voltage warning, runs Chromium, Kstars etc.

Could it be a defective Pi board?

rkaczorek, 2019-02-15:
Yeah, it looks like. That's why I mentioned a hardware problem. I would try to return it to a supplier, as a defective product.

closed 2019-02-15 by rkaczorek