rkaczorek/astroberry-server #14: VNC in every session?

Rigelsirius, 2019-02-15:
For connections by VNC, I need execute in every session VNCserver. I want to have the rapsberry in my personal remote observatory, and I don't know how can I would connect by VNCclient in my PC without to do in first time the manual activation of the VNC server in the Rapsberry.
Is it possible execute it automatically in the setup of the rapsberry session?
Sorry for my poor english, and congratulations for your excellent work in this astroberry server!!!

rkaczorek, 2019-02-15:
You don't need to do this. x11vnc server is autostarted on Astroberry Server at boot on standard port 5900.

closed 2019-02-15 by rkaczorek

Rigelsirius, 2019-02-18:
Thank you Radek. I was try, and it's ok!!!

rkaczorek, 2019-02-18:
I'm glad it works for you. Share your experience with the system on https://indilib.org/forum/astroberry.html