rkaczorek/astroberry-server #15: VAP and Home Wlan compatibility

Rigelsirius, 2019-02-15:
Hullo!
when I active my rapsberry pi 3 Model B, after SD flash and expand the space with raspi-config, VAP works correctly. But, when I configure my Home Wlan, VAP wireless does not appear. Is it incompatible? Can I do the remote control by ethernet lan connection?
Thanks!

rkaczorek, 2019-02-15:
This is normal. VAP autostarted at boot only if there is no connection to your home network.
When you connect to your home network VAP is deactivated and you can control Astroberry Server by accessing it in your home network. When you start Astroberry Server in the field where home network is not available, VAP will be styarted automatically and you can connect to it to control the system. You can also access the system over wired connection via ethernet port.

closed 2019-02-15 by rkaczorek

Rigelsirius, 2019-02-18:
Hi Radek,
Thank you very mutch for your help. You're a great person!!!

rkaczorek, 2019-02-18:
Thanks for your words. Clear skies to you!