rkaczorek/astroberry-server #16: No local WiFi and change keybord layout

HoudayerPierre, 2019-02-17:
Hi,

I have no WiFi connexion, edited the /etc/wpa_supplicant/wpa_supplicant.conf with ssid and psk (my wifi password) and reboot, no ip in ifconfig (http://astroberry.local/ => ERR_NAME_NOT_RESOLVED). 

And with the raspi-config, I can't change the keyboard layout.
With other raspi-config i do :
1. Run raspi-config.
2. Choose **Internationalization.**
3. Choose Keyboard setup.
4. Choose your keyboard or one of the generics.
5. Choose your langauge. For instance, for American English choose English (US)
6. sudo reboot.
But **Internationalization** is not visible.

rkaczorek, 2019-02-17:
Registering local name depends on your router & local network configuration. Check the router (wifi clients and dhcp clients) to see if rpi is connected to the network and what is the IP address. Then you can connect using http://IP/
Why don't you change the regional settings with GUI? System / Preferences / Hardware / Keyboard.

HoudayerPierre, 2019-02-17:
Yes i check my router but there is no ip for the rpi. I installed wpa_gui to see more things, and i can't do anything it's say "Could not get status from wpa_suppliant".
I also try "sudo wpa_suppliant -Dnl80211 -iwlan0 -c /etc/wpa_suppliant/wpa_suppliant.conf" 
And it say "you may have wpa_suppliant process already running" so it has to be good ... but no :(
(use usb wifi dongle too)
Ok thanks for the tip i never use the ubuntu mate thanks.

rkaczorek, 2019-02-17:
If your Raspberry does not connect to your WiFi router it might suggest that you are using Raspberry Pi 3B+ and your router does not support AC mode. Take a look at the last question in https://github.com/rkaczorek/astroberry-server#faq

HoudayerPierre, 2019-02-17:
No i'm using a raspberry pi 3 B (not +) but I don't want to use AC mode just to connect it on my local wifi network. 
Something is strange with my astroberry, with raspian and it's wifi gui, i can connect to my local wifi and get an ip (so the onborad wifi ship work and my router asign an ip). I switch the card for the one with astroberry and ... no wifi configured.
I run (apt-get) update, upgrade and dist-upgrade too but it not help.

HoudayerPierre, 2019-02-18:
I flashed a new card with astroberry did the same thing. rebooot few times, let it power off over night and this time get an ip from my router... 
Maybe a bad flash the first time. any way thanks for your amazing work :)
Did you plan to make usbip server in it (work with windows or macOS client) ?

HoudayerPierre, 2019-02-18:
Reopen because after I run (apt-get) update, upgrade and dist-upgrade it broke it again :(
Reflash but still no wifi ... no ideas ?

rkaczorek, 2019-02-18:
C'est bizzare ;) After fresh install you should see new astroberry wireless network. Connect to it and configure your system, connect to your home network, reboot. After reboot astroberry wireless network should disappear but your device should be accessible in your home network. Still the hostname is astroberry.local but IP should be assigned by your home router. You can check the ip by running "ping astroberry.local".
As I said before, update does not change the system configuration and should not influence wifi configuration.

HoudayerPierre, 2019-02-19:
Yes, this is strange but I miss one word, "after re-flash no **local** WiFi" sorry, The astroberry wireless network (AC mode) works I can see his SSID on my smartphone (and IP). 
The only thing is doesn't work is to get IP from my local WiFi network router, but if I register my local WiFi network SSID and PSK in astroberry, so not using AC (no strawberry SSID) ifconfig show wlan0 but with out IP from my router. 

What happen one time (when it works):
- I get IP one time after a night after rebooting a few times without working.
- I was happy, made upgrade, shutdown astroberry, connect it with usb to my telescope mount with usb ... blabla. 
- Finally (connect power and wait a little minute) I try http://astroberry.local and http://192.168.1.41 (see it before on astroberry, before upgrade and my router and i was able to ping it) but get "ERR_NAME_NOT_RESOLVED". 
- (Disconnect every thing except power it's a tiny UPS)
- Connect to HDMI and screen keyboard mouse. ifconfig show wlan0 but without IP from my router :(
- Re-flash a new card maybe upgrade broke something (now I know it's not thanks) reconfigure my local WiFi but nothing after few reboot no IP from router.

There is something here but I don't see what is wrong. 
(what did you have in your wpa_supplicant.conf)

closed 2019-02-19 by HoudayerPierre

rkaczorek, 2019-02-20:
Do you have uPnP active in your local router configuration?

HoudayerPierre, 2019-02-21:
Yes it's recommended by our ISP (Orange) in France for the TV (I think). (The DHCP pool size is big enough)
I just wait like before (power off the raspberry over the night and day) and the evening power up and it get an IP immediately. Something very strange ...

Can you, for the next release, make a save of your desktop customization because when i change it, it was not registered, so I lost it. (Not problem for me i want a full red and black, I make my own but maybe for the others)

rkaczorek, 2019-02-21:
I think this must be related to your local network configuration. Whatever I do I can't reproduce this issue. Can you test this on other router? e.g. activate hotspot on your mobile and configure astroberry to connect to it.
Also, desktop configuration is untouched in the system. It is exactly as in Ubuntu Mate. The only customization is a wallpaper and Astronomy apps menu.

HoudayerPierre, 2019-02-21:
I think too, always strange problems with ISP's routers ...
But is it working because the smartphone deal with it better or because is it working now at all ... i dont know. only thing is two time to wait over 24h and it's working now finger cross ;)

Ok, but when I entered in the "appearance menu" there is a "_custom_" one in italic... anyway,

Thanks for your work on this :D

rkaczorek, 2019-02-21:
I just thought you can test it with mobile hot spot to check if the issue is related to your router and local network. I have never had problems with IP lease from home router, USB 4G dongle or mobile. It just always worked.
I have just tested changing look & feel and apperance and it works as expected after reboot.

HoudayerPierre, 2019-02-22:
Yes, I was not clear, it's working with the smartphone.
Yes it's work great, it's just if I want use the old one "**custom**" of the fresh new install it's gone / replace by my changes. But you say it's the default one so if I use the one in here it's ok I thought you made some customization on it and not save one backup.