rkaczorek/astroberry-server #17: gphoto2 lib_usb_bulk_transfer timeout when capturing Nikon DSLR d7000

MaheshRaviVarma, 2019-02-27:
6.666016 ptp_usb_getdata             (2): Reading PTP_OC 0x1008 (Get object info) data...
6.666036 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
6.666151 gp_libusb1_read [libusb1.c:577](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Pipe error (-9)
6.666185 gp_port_read [gphoto2-port.c:437](0): Reading 1024 = 0x400 bytes from port failed: Error reading from the port (-34)
6.666203 ptp_usb_getpacket           (2): Clearing halt on IN EP and retrying once.
6.666221 gp_port_usb_clear_halt      (2): Clear USB halt...
6.666282 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...

MaheshRaviVarma, 2019-02-27:
If I do "tail -f <log_file>" I get the above error printed for some time and then a huge amount of data starts getting written (after say 20 seconds or so)

MaheshRaviVarma, 2019-02-27:
Command used:
$ gphoto2 --auto-detect --set-config=/main/capturesettings/shutterspeed=43 --debug --debug-logfile=./mrv.log
$ gphoto2 --auto-detect --capture-image-and-download --debug --debug-logfile=./mrv.log
$ tail -f ./mrv.log

MaheshRaviVarma, 2019-02-27:
The error occurs only if a capture of more than (say) 2-3 seconds is triggered !

rkaczorek, 2019-02-27:
I can't see how it is related to Astroberry Server and not gphoto2. There is no change to standard gphoto2 tool in Astroberry Server, which is built on Ubuntu Mate 16.04. Please file a bug at gphoto2 or Ubuntu Mate if you think it's a bug.

closed 2019-02-27 by rkaczorek