rkaczorek/astroberry-server #18: VNC is very laggy

MaheshRaviVarma, 2019-02-27:
Trying to connect to RPi over VNC - it is very slow and laggy. The response time is in the order of seconds

MaheshRaviVarma, 2019-02-27:
This is especially so - when I try to remote connect over VNC to Kstars running on RPi

rkaczorek, 2019-02-27:
Please provide more details. Are you using builtin VNC server (x11vnc)? How do you connect to VNC server (what client, wire or wireless, configuration changes?). What is your Raspberry Pi version?

MaheshRaviVarma, 2019-02-28:
* I didn't start anything explicitly, that means it is the builtin one (x11vnc)
* wireless connectivity over 'astroberry' VAP
* RPi version: 3B

Let me know, if you need more details. Thank You

MaheshRaviVarma, 2019-02-28:
* Client: I tried both the 'browser' (Mac Safari and 'VNC viewer' (MAC built in VNC client

rkaczorek, 2019-02-28:
Well, so everything seems normal. Can you confirm that you're not a victim of undervoltage of Raspberry or CPU overheating? How much ampers your PSU provides for Raspberry? What is a temperature of the CPU? In both cases CPU might scale down.

MaheshRaviVarma, 2019-03-01:
How do I confirm these ?
- under voltage ?
- over heating ?

and if yes, then how would I correct them ?

rkaczorek, 2019-03-01:
To check temperature run
`/opt/vc/bin/vcgencmd measure_temp`
To check undervoltage connect Raspberry to HDMI and check if yellow lightning is appearing in the top-right corner of your screen.

closed 2019-05-06 by rkaczorek

davthomaspilot, 2022-11-25:
Maybe I should start a new thread, but I have a similar issue, but only after some reboots.

Performance under x11vnc is so bad, the echoes of keystrokes take seconds.  Meanwhile, an ssh connection has no perceivable delay while typing.

Running "top" shows no process consuming much CPU or memory.

Ethernet connection between the rpi and router, same with desktop that's running the vnc client.

Many (most?) times after reboot, everything is fine--great performance using lxsession, just like ssh.  

File system is no where near full.

dmesg does reveal anything, at least that I can realize.

Temperature is fine.  No undervoltage lightning bolt.  

I'm running out of ideas for debugging.  I'd appreciate any suggestions.