rkaczorek/astroberry-server #20: Unable to connect from Safari browser

wornish, 2019-03-07:
Loaded the latest version of astroberry today.

ssh connection works fine. via Wifi
VNC connection also works fine but is a little laggy.
When I try and connect via my browser I get this error message

[image]

rkaczorek, 2019-03-07:
javascripts are not loading properly - just refresh the page until it disappears, you should be fine

wornish, 2019-03-08:
Thanks for getting back. 

I tried it from my Windows 10 laptop and it works fine, which is where I will be using it anyway, so that’s good.

I have now  tried multiple times from Safari on my iMac and I still get the same error.
Seems like its something strange going on with that.  The connect window appears and it lets me enter my password then when I click connect I get the error message.

  I have played with EKOS and it looks promising.

Regards
Dave

wornish, 2019-03-08:
It's now working on Safari, and on Windows 10.   Seems the 5V supply to the RPi was marginally low.  I have changed the way its connected and all now working as it should.

Sorry for the false alarm.

closed 2019-03-08 by wornish