rkaczorek/astroberry-server #21: Astroberry server does not fall back to AP (access point) mode when no WiFi is available

nikivan1, 2019-03-07:
After configuring access to a home network, Astroberry never falls back to AP mode, even when the network is no longer available. Tested by turning off the router power supply, and also on a field trip.

HoudayerPierre, 2019-03-08:
Try this : (in the FAQ)

Q: I can't see astroberry virtual access point on my Raspberry Pi 3B+

A: It is there! The problem is that your PC does not support wireless ac mode. You need to connect HDMI screen, login and change two parameters in /etc/hostapd/hostapd_3bplus.conf to: hw_mode=g and channel=6.

nikivan1, 2019-03-08:
My PC supports wireless AC mode. After a fresh install, I am able to connect to the astroberry access point. But after configuring a connection to my home WiFi, astroberry never falls back to access point mode, even when WiFi is unavailable (router is powered down, or on a remote location). I thought I was pretty clear about that in my initial post.

iNFest84, 2019-03-09:
I have same issue with vap on Pi 3B.

rkaczorek, 2019-03-09:
What's the result of running this? `sudo astroberry_vap start`

iNFest84, 2019-03-10:
```
sudo astroberry_vap status
Virtual Access Point STOPPED
sudo astroberry_vap start
Virtual Access Point STARTED
```
Ap is not present until i remove information about my my home WiFi /etc/wpa_supplicant/wpa_supplicant.conf

nikivan1, 2019-03-12:
In my case running 'sudo astroberry_vap status' does not show anything.

jimwhitelaw, 2019-10-10:
I'm seeing this issue too. Downloaded IMG last week. AP does not start up if the local wifi is turned off. While testing this, I also noticed that the astroberry doesn't reconnect to the home network after wifi is re-enabled. I'm using a RPi 3B+.

rkaczorek, 2019-11-18:
Removed in v2.0.0

closed 2019-11-18 by rkaczorek