rkaczorek/astroberry-server #28: Is it possible to install the atik air program?

aday5, 2019-06-17:
Is it possible to install the atik air program?
I use a ccd atik and I would like to install the artemis capture program. it's possible?
Thank you

rkaczorek, 2019-06-17:
No. You cannot install it as an additional component. Atik provides full system image for Atik Air, which you can use to flash your SD card.

closed 2019-06-17 by rkaczorek

bruceparker, 2020-11-08:
Interesting.  Page 8 of the advanced section describes how to download and install he components.  Whether it still works is another matter, but now I'm curious also.

http://www.atik-cameras.com/wp-content/uploads/2016/09/AtikAir-Guide.pdf

rkaczorek, 2020-11-09:
Oh, so it looks like it might work. Haven't tried it but seems ok to install it as additional component.