rkaczorek/astroberry-server #30: Bluetooth timeout

gvandela, 2019-07-14:
I installed a bluetooth serial port module in my mount and this was working flawless with INDI on a raspberry pi zero w running raspbian. However, using the astroberry image, I get timeout errors and I cannot get my mount to do anything.
I'm having a hard time finding any info on this issue and I'm stuck. Is this a known issue with the particular version of Ubuntu Mate? Any suggestions on what I could try next?

gvandela, 2019-07-14:
I started from scratch with Ubuntu Mate 18.04 and got my mount to respond, no more timeout errors. Maybe it with the Ubuntu Mate version used on the astroberry image?

rkaczorek, 2019-07-15:
Version 1.1.0 is based on Ubuntu Mate 16.04 Desktop. Ubuntu Mate 18.04 is still beta release. I'm still waiting on the stable version. What version of Raspberry Pi are you using? By any chance is it 3B+? If so I would not be surprised with BT problems. It work for me, but for many people doesn't.

gvandela, 2019-07-15:
I am indeed using a 3B+. Thank you for your confirmation!

closed 2019-07-15 by gvandela