rkaczorek/astroberry-server #33: Kstars not showing profiles created by Astroberry and other issues.

cstachon, 2019-11-21:
Hi Radek,

As requested info on above problem.

1. Existing Kstars/Indi was installed under PI user

2.Astroberry installed as per instructions using APT etc

3. Astroberry is able to see only User Astroberry profiles (guess thats what is expected)

4. Created new profiles via Web Browser utility - these can be seen even after power off/on

5. Running Kstars via Menu - education etc  via REALVNC viewer - ekos only see's original pre Astroberry profiles.

6. Astroberry Connect does NOT work with my Realvnc server - which was installed before Astroberry but is up to date. Press "connect" in Web browser and just displays "failed to connect". Thats using either Hotspot or Static Wired Ethernet addresses. Dont get the option to input a password!  Realvnc viewer works as before Astroberry install.

No DOT(.)indi directory exists under Astroberry user but does still exist under PI.. Used ls -la to verify which should have shown "hidden directories".

Guess problem is with having existing user PI with Indi/Kstars/ekos compiled and installed,

Where are the log files for Astroberry server etc kept - nothing shown in SYSLOG,DMESG etc

Also Rpi-config is not holding on to changes for Wifi (GB).

Hotspot working fine even when connected to static wired ethernet - i.e. both working together.

Weather plugin does not hold onto location change using UK post code - it might be warmer in Warsaw but not really helpful. Reverts back to Warsaw when HOTSPOT only connected.

To be fair I would have used a clean install but haven't a spare SD card at the moment

cstachon, 2019-11-21:
some pics

cstachon, 2019-11-21:
https://drive.google.com/open?id=1fp7Lpy9WC-Fd675L3WQOISOqdqHdNotR

pics of things

cstachon, 2019-11-21:
some more info - found a spare sd card and used the Astroberry image.

1. Now can connect vnc any way I want - even realvnc viewer works with default password
2. Weather Ap still doesn't safe preferences - used both sudo raspi-config and menu-preferences still comes back to Warsaw after reboot.  Note Weather plugin doesn't disappear when pressing OK button. 
3. Ok profiles via Web browser are separate from Kstars as I expect that you are assuming all local or remote when using browser. Would be nice to have same profiles.
4. GPS seems flacky - get a lot of bad gateway messages when pressing GPS button in Web browser

So it looks like updating an existing Kstars/PI set up causes problems especially with remote desktop via Web Browser (NOVNC ?)

cstachon, 2019-11-22:
Some more info which may or not help. Doing a direct call to the Web manager API doesn't show the Kstars/Ekos profiles just the Astroberry ones like this [{"id": 1, "name": "Simulators", "port": 7624, "autostart": 0, "autoconnect": 0}, {"id": 2, "name": "stash1", "port": 7624, "autostart": 1, "autoconnect": 1}]

rkaczorek, 2019-11-22:
There are a few issues here.

Weather Plugin is part of Taskbar and it behaves strangely at the first change. If you set the location for the first time use Panel Settings / Panel Applets and scroll to Weather Plugin, then click Preferences. See #35 for details.

I need to investigate on web-indi profiles. What you're saying is that KStars sees other profiles that set from web-indi?

RealVNC - you probably need to review your VNC settings. Astroberry uses service mode, which does not use user level configuration. I suspect that yours runs in user mode and expects configuration in astroberry/.vnc directory.

Are you runnig  **sudo** rpi-config? No changes introduced to this component whatsoever.

If you get bad gateway for GPS run sudo systemctl restart gpspanel.service - it looks like there is a problem with it. Do you have gpsd installed on your system?

Astroberry logs are in /var/log/astroberry

cstachon, 2019-11-22:
Weather plugin - solved for now thanks.

Web-indi profiles - I expected to see the same profiles in both Kstars and Astroberry side panel. They each shown there own but not the other. So If I create a profile in Astroberry Side panel it does not show in Kstars and any profiles I create in Kstars do not show in Astroberry.  Question is should I be able to see profiles created in both ?

Realvnc -  a red herring - it works as does Novnc - Realvnc just doesn't see the sidepanel which is correct ! I can run both at the same time and see the same application loaded no matter which loads the application. Not something I would do normally :-)

Yes I have a Working GPS dongle - although testing indoors doesn't help Sat pickup .

No I ran sudo raspi-config   not rpi-config - bad typo by me !

rkaczorek, 2019-11-23:
- I'm glad Weather Plugin works. Please continue this in #35 

- Web-indi profiles - I have just tested it and you can create profiles for web-indi from Ekos. When creating profile you need to choose Remote Host AND select INDI Web Manager tick box. After saving profile it is available in web-indi. However it does not refresh automatically so you need to restart web-indi (run: sudo systemctl restart indiwebmanager.service). It does not work the other way around. Profiles created in web-indi are not automatically available in Ekos. This This is how Ekos and web-indi are designed. To change this behavior one need to file request for change at these projects sites https://github.com/knro/indiwebmanager and https://github.com/KDE/kstars
BTW, if you want to migrate your profiles just copy .indi/profiles.db to a user account you are going to use.

- RealVNC - this is right sidepanel is available **only** when you access the system with a browser.

cstachon, 2019-11-23:
Web-indi profiles - I had already tried this except I rebooted the system so Indi Web Manager would have restarted. I will try again and see what I get. Is not a big deal so long as you know thats how it works :-)

cstachon, 2019-11-23:
Tried it again and yes it worked - dont know why it didn't show before - finger trouble maybe.

I might ask for a change as it gives my flexible options of trouble shooting my normal remote CCDciel working and a "local" Kstars/Ekos backup. So I will close this :-)

closed 2019-11-23 by cstachon