rkaczorek/astroberry-server #35: Astroberry Panel shows default Warsaw after GPS has made a fix

cstachon, 2019-11-22:
Weather plugin was changed to show my location ok - only works if no reboot as it goes back to Warsaw.
However the Celestial applet of the side panel still shows Warsaw even though GPS had made a fix and I had changed the weather app to correct location.

It would be useful to have one location source so that all apps report the same location ! 

Unless there is a specific way to do this which I have missed.
Not a major problem!

rkaczorek, 2019-11-22:
Astro Panel (accessible in the sliding menu) determines location in the following order:
- **user weather file** (/home/astroberry/.config/lxpanel/LXDE-astroberry/panels/panel)
- **system weather file** (/etc/xdg/lxpanel/LXDE-astroberry/panels/panel)
- **config file** (/etc/astropanel.conf)
- GPS

At the first boot only **system weather file** exists, so default location for weather and Astro Panel are Warsaw, Poland. As soon as you customize your Taskbar **user weather file** is used. BTW If you set the location for the first time use Panel Settings / Panel Applets and scroll to Weather Plugin, then click Preferences. Otherwise the settings will not be saved. It should not be a problem for subsequent changes. This comes from the panels logic (I do not program this part) - the first customization of Taskbar makes a copy of **system weather file** (default) to **user weather file**.
Anyway, when changing Weather Plugin location you need to be online, because the Weather Plugin uses Open Weather Map as source of location and weather data.

Nevertheless, you can create your own configuration (see **config file** above). This should make Astro Panel ignore Weather Plugin location data. It should but it does not at the moment due to my mistake. As the matter of fact the proper order should be:
- config file (/etc/astropanel.conf)
- GPS
- user weather file (/home/astroberry/.config/lxpanel/LXDE-astroberry/panels/panel)
- system weather file (/etc/xdg/lxpanel/LXDE-astroberry/panels/panel)

I realized this after I have released the system. It will be changed in the next version. It will come with regular system update.

Could you confirm this behavior?

cstachon, 2019-11-22:
Strange reboots didn't save weather panel changes but I have done nothing since I changed the location as i stated above BUT i did close down and start up again after power off.  Now both Celestial & Weather are correctly set. So for now its working. I will wait until next version and see how it goes then. Thanks Radek it does look good - well done.

rkaczorek, 2019-11-22:
Could you check is the file /home/astroberry/.config/lxpanel/LXDE-astroberry/panels/panel exists now?

cstachon, 2019-11-23:
Yes I can confirm that the file/directory structure exist - :-)
So Close ?

closed 2019-11-23 by cstachon

rkaczorek, 2019-11-23:
Fixed in v2.0.1 - please update

cstachon, 2019-11-24:
Sorry its started to go wrong again - weather back to Warsaw.  If GPS fails (which it does if your reboot but is OK if power off then power on) then weather plugin shows Warsaw again. You can change it via settings but it only lasts until next boot or reboot.  I had NOT changed the plugin details  - honest.  However I then changed all the files as suggested and used my nearest known city and it started to work again.   Must be a better way but I dont know what!

So I think it will now work as follows :-

1. If GPS working and 2d/3d fix it should pick up and use GPS location

2. If GPS not working , cant get a fix OR at my base Obsys it will default (i hope) to one of the 3 config files which contain the same co-ordinates

Perhaps not so much of a bug as a messy way to do things IMHO :-)

Cant we have ONE file(no permissions set - so access all ) for defaults which is parsed on boot up to create the panel files required ?

cstachon, 2019-11-24:
Have done a few tests with and with GPS and doing as I have put in my last comment seems to work ok. Only minor point is to change comment in slider panel to say GPS and Celestial not working not 502 BAD GATEWAY and the HOTPLUG of GPS and restart service nginx still gives the same error - you seem to have to reboot! But that is a future enhancement one day when you are bored LOL

rkaczorek, 2019-11-24:
Take look at the last FAQ. You can use /etc/astropanel.conf to enable/disable GPS.
If GPS is enabled but no fix obtained Astro Panel falls back to static location set in this file. It loops the check every 60 seconds so eventually it gets the GPS fix.
The only case it uses Weather Plugin location data is if the config file does not exist at all.
The ultimate fall back is demo location, hard coded in Astro Panel, activated if no GPS, no config, no Weather Plugin files are available.

Bad gateway is separate issue. It comes only when Astro Panel service is down for any reason. 
Please debug this case by:
1) edit /var/www/astropanel/astropanel.py and set DEBUG = True at the top of the file
2) stop the service by running sudo systemctl stop astropanel.service
3) run panel in terminal: python3 /var/www/astropanel/astropanel.py
4) Observe the output and conditions it exits

The system is using nginx proxy, which delivers astropanel (running on port 8626) on port 80 or 443. That is why if astropanel service is down, nginx reports bad gateway (upstream service not available for nginx proxy).

cstachon, 2019-11-24:
I can only report what happened - without alteration to any files and after the update/upgrade it worked most of yesterday no matter what GPS was doing (I even unplugged it during my testing) . Always showed same details or expected details.  Power up/down it worked, Reboot it worked. This morning it worked. Then it reverted back to Warsaw on the plugin.  I fully acccept what you say and did the changes in the FAQ BEFORE reopenning the problem on here.  Will just wait and see over the next few days.

Bad gateway I will do the debug and report back.

cstachon, 2019-11-25:
OK did as you asked

Changed line in py from "debug = False" to "debug = True"
Stopped service
ran python from command line
still showed "bad gateway" when clicking on GPS on Astropanel
Nothing output from erminal
So I refreshed the screen which meant connecting again and it produced this
Loading values from configuration file
/etc/astropanel.conf
53.03805
-0.457532
62
Cranwell
Cranwell
Loading values from configuration file
/etc/astropanel.conf
53.03805
-0.457532
62
Cranwell
Cranwell
^CKeyboardInterrupt
2019-11-25T14:05:16Z
Keyboard interrupt received
Terminated by user
Good Bye.
============
NOTE Bad Gateway message gone and GPS MAP etc working - so no real help I guess !

rkaczorek, 2019-11-25:
If you get bad gateway on GPS Panel, you should do similar procedure with this service.
1. Stop GPS Panel service: sudo systemctl stop gpspanel.service
2. Run panel in terminal: python3 /var/www/gpspanel/gpspanel.py
3. Observe the output and conditions it exits
There is no specific debug mode but if the panel exits for any reason it will render 'bad gateway' error.

rkaczorek, 2019-12-01:
The update has been released. From now on GPS Panel and Astro Panel use only GPS readings as source of location. For users that do not have GPS, the system provides Virtual GPS, which can be configured by editing /etc/location.conf file or using Preferences/Geographic Location menu.
The /etc/astropanel.conf file has been abandoned and is not used anymore.