rkaczorek/astroberry-server #36: tablet connection

scopewizard2019, 2019-11-23:
I can connect access point with laptop (Windows 10) using VNC Viewer but not with my tablet A (android). MSG: The computer's domain name could not be resolve.

rkaczorek, 2019-11-23:
It happens sometimes and is device dependent. In such a case determine IP address and use it instead astroberry.local name. You can do it for example running: ping astroberry.local from a device that sees the domain name

scopewizard2019, 2019-11-25:
Ok, I got it to work on Android Tablet. Instead of the ip address (hotspot
ip 10.42.0.20 port 5900), I entered (10.42.0.1 port 5900) and got access
just fine.

Denis

rkaczorek, 2019-11-25:
You can do the same pointing your web browser to 10.42.0.1

closed 2019-11-25 by rkaczorek