rkaczorek/astroberry-server #37: loading of raw (cr2) files fail on linux

cstachon, 2019-11-25:
Astroberry by default does not have a RAW decoder it seems so loading CR2 files fail in ASTAP/CCDCIEL. Required ,for me at least, as I pass across CR2 files I have taken from APT(windows) across to ccdciel which as I run on Windows normally has the RAW file decoder preloaded in the install of CCDCIEL.  Not the same for Linux !

ccdciel reported this 
21:27:57 Could not find libpasraw.so.1, libraw.so or dcraw. Loading camera raw files is not possible.
21:28:16 No RAW decoder found!
ASTAP reported this 
Could not find program unprocessed_raw !!, Install libraw. Eg: sudo apt-get install libraw-bin

cstachon, 2019-12-03:
If you install Libraw CR2 files are loaded and processed ok in ASTAP - unfortunately this doesn't work for CCDCIEL (4.65)

jscheidtmann, 2021-02-14:
Did you install the libpasraw library, too? This is the pascal wrapper for libraw (Ccdciel is a Pascal program).