rkaczorek/astroberry-server #38: GPS possible problem

cstachon, 2019-11-26:
Not sure if this is hardware related or python but GPS is not working well with Astroberry Panel at times.

Symptom:-

1.  Panel says "waiting" for GPS fix.
2. If you run CGPS this clearly shows GPS is working and displays correct details.
3. You can wait over the 60 secs refresh time (I tried 5 mins) and the situation as described in (1) still exists and CGPS is still working.
4. Reboot and both GPS in Astropanel and CGPS are correct and sync'd.

Debug just says "trying GPS ........ " then "no coordinates" and takes the default

Ublox-7 is my GPS dongle but the above problem doesn't have a pattern. As I have said before the GPS is next to the Window so not best placed but the fact hat CGPS is working ok seems to suggest a problem with Astropanel GPS/Python program. I do wait a good deal of time for the GPS to stabilise.
 

Sorry I could find more info !

rkaczorek, 2019-11-26:
Do you hotplug GPS dongle while the system is up and running? or it works this way even though GPS dongle is connected at boot?

cstachon, 2019-11-26:
normally I just leave it plugged it.  Just wondering if it a power problem so i have attached a powered hub and plugged in the gps cable to that. See how that goes!

cstachon, 2019-11-27:
Some more info :-

I believe we have 2 issues and one if not Astroberry its Ublox/GPSD.

I GPS Panel isn't working you cannot unplug the device and plug it back in or even restart (or stop and start) cgspanel.service as you get the following in DMESG  cdc_acm 1-1.2.1:1.0: failed to set dtr/rts .
Under these conditions it will never work and the only way to restart GPSD and get gpspanel working is to
1. Stop GPSD - I did a kill not a restart service
2.stop gpspanel.service
3. Unplug GPS and plug GPS back in
4. GPSD will start automatically as Hot Plug
5. start gpspanel.service
6. normally then ok.

However I believe there is something else 

While running cgps to check if GPS ok and pressing gps button CGPS stops  working and gpspanel doesn't work either.    Is the python module for GPS running as a std "client" ?   Trouble is cant repeat this problem every time.

systemctl status gpspanel.service reports nothing wrong e.g. 

gpspanel.service - GPS Panel
   Loaded: loaded (/etc/systemd/system/gpspanel.service; enabled; vendor preset:
   Active: active (running) since Wed 2019-11-27 13:03:55 GMT; 13min ago
 Main PID: 1843 (python3)
    Tasks: 2 (limit: 2077)
   Memory: 24.2M
   CGroup: /system.slice/gpspanel.service
           `-1843 /usr/bin/python3 /var/www/gpspanel/gpspanel.py

The DTR/RTS problem with Ublox seems to be common if I do a google search but I have not found any helpful ways to correct it

cstachon, 2019-11-27:
Sorry Radek some more info
Sometimes the GPS Map stops being shown even though there is a fix - just a blank screen where the map should be.
However if you do a manual browser to <ip>:8625 this shows the map as it should - strange :-(

Might also have a power supply problem on RPI3b+ as I get a lot of "under voltage" messages in DMESG even though I supply 10amp 5v - this might also account for ublox problems.  I am going to ask some of the Astroberry users on SGL to check there DMESG log.

rkaczorek, 2019-11-27:
gpspanel.py is a rgular gpsd client, based on gps3 library for Python3. It must be an issue with the library then. I used to use Python2 library in the past, however Python2 is reaching end of life January next year so I ported it to Python3. Let's live with it for a while and we'll see where it goes.

cstachon, 2019-11-27:
ok

rkaczorek, 2019-11-30:
All the gpspanel components have been updated (openlayers, jquery, bootstrap). Please test if it works ok

cstachon, 2019-12-03:
seems ok

closed 2019-12-03 by cstachon