rkaczorek/astroberry-server #41: Feature list incorrect on readme.

setnes, 2019-11-30:
Is the list of features correct on the readme page?  I don't see some of this stuff and am wondering if I missed something when I installed from an existing Raspbian build.

I don't see things like Cartes du Ciel and CCDciel, but I do see Kstars, PHD2, and oaCapture.  I might not need the missing stuff, just curious if the readme is up-to-date.

Also, do you recommend getting a USB GPS device for the Raspberry Pi?  It would be nice if I could rely on the polar finder chart.  Without an accurate location, I think it's going to be incorrect all of the time.

rkaczorek, 2019-11-30:
Added "(only in precooked image)" tag. All of these are available from their maintainers' repositories and can be installed separately. As stated on the astroberry.io site installing system on your own is for advanced users who want to manually configure all the system details.

Regarding Astro Panel, already discussed in #35 - you can configure your static location in /etc/astropanel.conf

closed 2019-11-30 by rkaczorek

setnes, 2019-12-01:
Thanks for clarifying what is only in the "precooked" image.  I installed it from your image instead of a Raspbian image to see what was different.  

I think the changes you are tweaking for providing a virtual GPS might help with my concerns.  The weather plugin only works if you have an internet connection.

Thanks for putting this all together.