rkaczorek/astroberry-server #44: Astropanel GPS stuck to Warsaw

blemasle, 2019-11-30:
Hi,

On a fresh and upgraded astroberry install, the GPS location cannot be forced using [configuration files](https://github.com/rkaczorek/astroberry-server/issues/35#issuecomment-557670123).

After setting my location coordinates in `/etc/astropanel.conf` with `use_gps = no` and enabling [debug ouput](https://github.com/rkaczorek/astroberry-server/issues/35#issuecomment-557911808), I can cleary see `astropanel.py` using my configuration file

```
Loading values from configuration file
/etc/astropanel.conf
xxx
xxx
0
xxx
My location
```
But somehow, the astropanel is locked to Warsaw and is waiting for a gps fix
[image]

rkaczorek, 2019-11-30:
What you are referring to is **GPS Panel**, not Astro Panel. This is designed to show real GPS position and if you don't use GPS it defaults to Warsaw. It does not use /etc/astropanel.conf configuration nor Weather Plugin configuration. This behavior will be changed in the future versions.

blemasle, 2019-11-30:
You're absolutely right, sorry for the confusion. I naturally reffered to that map to ensure that GPS coordinates used in the astro panel were correct. The change you're talking about makes perfect sense from a "dumb" (as in me) user point of view 👍 

Thanks !

rkaczorek, 2019-11-30:
Your input is much appreciated. It brings to my attention the fact that the configuration is a bit confusing. I will be releasing a solution for this soon

rkaczorek, 2019-12-01:
The update has been released. From now on GPS Panel and Astro Panel use only GPS readings as source of location. For users that do not have GPS, the system provides Virtual GPS, whch can be configured by editing /etc/location.conf file or using Preferences/Geographic Location menu.
The /etc/astropanel.conf file has been abandoned and is not used anymore.

closed 2019-12-01 by rkaczorek

blemasle, 2019-12-01:
That seems perfect, thanks !