rkaczorek/astroberry-server #47: GPS (latest version) causes problem

cstachon, 2019-12-01:
Radek, 

Having a real GPS switched on makes the map repeatedly jump between default location and the fix from the real GPS.  Ok if no real GPS attached.

rkaczorek, 2019-12-01:
If you use real GPS you need to disable virtual GPS.
There are at least two ways to handle this:
(1) uninstall virtualgps (sudo apt remove virtualgps),
(2) stop virtualgps and disable the service so it does not start after reboot (sudo systemctl stop virtualgps.service && sudo systemctl disable virtualgps.service).
You can enable or start it anytime by running: sudo systemctl enable/start virtualgps.service

There is also runtime control of what GPS device used by gpsd, very handy:
sudo gpsdctl add/remove device
(e.g. to connect/disconnect virtualgps in real time you run sudo gpsdctl add/remove /tmp/vgps)

cstachon, 2019-12-01:
Ok I can also disable it via the command line but it would be nice for less command line stuff - as someone said :-)  I and I guess others use it at home and in the field.

So another Enhancement maybe

closed 2019-12-03 by cstachon

setnes, 2020-02-23:
I know this is closed, but if someone happens upon this issue they need to know that option 1 for disabling virtualgps (sudo apt remove virtualgps) no longer works.  If you attempt to remove virtualgps, apt will try to remove the rest of astroberry.  Option 2 (disabling the service) is the way to go.

Also, if you had previously uninstalled virtualgps, and it quietly came back after an apt update, it does not play nice.  I might open a new issue to explain this.