rkaczorek/astroberry-server #51: Astrometry Index files

troske, 2019-12-13:
I am setting up Astroberry for the first time. I have used StellarMate.

When I go to settings to download astrometry index files, it says “downloads disabled.” The user interface is different from StellarMate, so I am stuck. I don’t see any check box to change the setting.

I see that the astrometry.cfg file is editable, but I am not a programmer.

How do I enable downloads so I can install the plate solving files?

I see reference to ASTAP, which I have heard of, but again, no idea how to set it up with index files for plate solving. Is there a manual for Astroberry?

My hope is to get this Raspberry Pi 4 working once and for all without all the crashes and headaches and workarounds of StellarMate.

Thank you for putting this package together. Looking forward to trying it under the stars.

Tim

porkyhttp, 2019-12-13:
Just put the FITS index files into /usr/share/astrometry

rkaczorek, 2019-12-13:
Click options on Align tab and choose .indi drirectory as a destination for index files

troske, 2019-12-13:
Thank you. I was able to switch to one of the directories and download the files.

troske, 2020-01-21:
So we finally had a clear night. Plate solving failed every time with astrometry.net, both online and offline. Offline, I receive an error message "Plate solve failed. Try again." Online, the image uploads, but the solver times out. 

I downloaded and installed ASTAP using a Debian package from the ASTAP site. I also downloaded and installed the G17 star index package. When installing ASTAP, I got an error saying not all components could be installed. In the alignment options, I have placed the correct path to the ASTAP.exe. The solver fails every time with "Solver failed. Try again."

Images were captured with two different scopes and cameras. ZWO120mm Mini on an Orion 50mm guidescope and a ZWO1600mm on the Rokinon 135mm f/2.  The field of view appears correct in the alignment tab. I am using the telescope simulator as my mount. Perhaps my field of view is too wide?

By the way, I am unable to copy any files into usr/share/astrometry because of a permissions error. The files are in another folder that is referenced in astrometry.config and their check boxes are marked when I go to options/index files. 

Is there some setting or checkbox in the alignment options that I am missing? I never had plate solve issues with Stellarmate.  I am going on one year of fighting the RPi EKOS system and am getting ready to give up.

rkaczorek, 2020-01-21:
Don't give up, be specific about the issue, by saying 'it's correct' does not let me verify any real conditions:
- what system are you running KStars on?
- what aperture and focal length is set in telescope simulator for scope and guider?
- what is the path to ASTAP?
- what is the calculated field of view?
- what indexes are downloaded and where are they located?

BTW. You cannot copy files to /usr/share/astrometry without root permissions - this is because write permissions of this directory. You don't need to place the files there. Optional (default) location is /home/astroberry/.local/share/kstars/astrometry

troske, 2020-02-01:
Just to let you know, I was able to get ASTAP working. Turns out that the Astroberry default path was not where ASTAP was actually installed. I still could not get it to do a blind solve and it would not solve when my field of view was far from the coordinates reported by the telescope simulator.

But with the simulator slewed closer to my actual view, it was extremely fast. I am using a non-Go To Star Adventurer mount. I plate solve and then verify what my camera is seeing with the SkySafari aux. Then I can move the mount manually to get on target.

jnowat, 2020-05-24:
Hey, mind if I chime in here and just request a simple line-by-line instruction on how to install all astronometry.net index files on astroberry so that they work with default settings in KStars? I'm trying to remote solve (kind of important on a rare, clear night like tonight when the astronometry.net server goes berserk and disables login...). I'm inside with my PC on KStars, watching all sorts of photos for plate-solving come in and fail. Did try to VNC, but I'm trying to use the Pi as a server, not a tunnel. I know I've installed index files in the past, but I can't seem to find the instructions now.

Please, for the world to see: line-by-line terminal commands to install all index files on RaspberryPi4 running latest Astroberry so that it may solve remotely and offline. 

This isn't on the Wiki for some reason, even though it states "Astrometry for field solving (index files not provided and must be downloaded separately)"... so maybe place detailed instructions there as well? _That_ would be awesome. I have plenty other issues with mount and downloading images that need to be addressed and I really can't troubleshoot plate solving issues and still have time for other things- I have maybe one more clear night before the moonlight rolls in- 4-5hrs of clear skies tonight that could have been better spent. Not enthused over here in north-coastal California. Astroberry worked "out of the box" (that's impressive to me).... until it didn't.

rkaczorek, 2020-05-24:
Step by step as you requested:
- Start KStars and Ekos
- Go to Align tab in Ekos
- Click Options in bottom-right
- Go To Index Files
- Select locations for index files in drop-down menu called Index Files Location. It should be set to /home/astroberry/.local/share/kstars/astrometry
- Select all required index files - they will be downloaded automatically for you. Note size of the files and be patient
- Enjoy!

Let me know if it works for you.

rkaczorek, 2020-05-26:
Any feedback?

jnowat, 2020-05-27:
I think I had written out a reply that was 1) quite long and 2) a little whiny in regards to a wretched experience I had one night, and representing the hair-pulling frustration I was experiencing- seems the response didn't post. It seems I can take dark frames now during these cloudy nights, so I've got that going for me. 

Let me reiterate that I am connected to Astroberry on Pi4 as server, using Win10 KStars as client. 

You said, "Select locations for index files in drop-down menu called Index Files Location. It should be set to /home/astroberry/.local/share/kstars/astrometry"

There is no drop down on my client? Further, I honestly can't find how to install the full set of index files through the terminal, which I wish to do on the Astroberry. I've done it before, but I can't wrap my head around how to get the index files downloaded. Completely lost in that respect :/ thought I was better with Linux than this at least

rkaczorek, 2020-05-27:
It's quite important thing to note that you use KStars on Windows 10 indeed. However, if so, your question should be "how to install astrometry on Windows 10", instead "how to install all astronometry.net index files on astroberry". The answer to the former question is, astrometry is not supported on Windows. The answer to the latter question has been already provided above.
Also please note that I do not provide support for KStars and Windows here. It is Astroberry Server project. If you follow my step-by-step instructions, use KStars running on Raspberry Pi and connect to it either via web browser or VNC, you will find out there is no issue at all.

jnowat, 2020-05-27:
*How do I download all astrometry index files in the terminal in the raspberry pi?*

rkaczorek, 2020-05-27:
> _How do I download all astrometry index files in the terminal in the raspberry pi?_
`sudo apt install astrometry-data-2mass astrometry-data-tycho2`
This applies to any linux system using APT software management system.
After running this all files (>30GB) are available in /usr/share/astrometry directory.

rkaczorek, 2020-05-27:
I should have noted that generic solution is described on [astrometry.net site](http://astrometry.net/doc/readme.html)

len5771, 2021-04-03:
Recently reflashed and reinstalled astroberry on one of my Pi's. Tried to download the astrometry files and only the Tycho2 catalogue was accessible. Unable to get to the 2mass ones. Downloaded via the 'sudo apt install astrometry-data-2mass' but really don't need the 33GB that that entails. Tried the same on my other astroberry install and that behaved same way, this one already has the required files so was no need to download them all. Just can't add any extra 2mass ones.

len5771, 2021-04-04:
Last nights session didn't go well. Looks as though the downloaded astrometry files weren't accessible anyway. Able to polar align and plate solve with my guide cam but not my main camera. Noted that there is a config file in with Tycho2 files but not with 2mass? Complete newbe to linux so haven't got a clue how to proceed.

len5771, 2021-04-04:
Just visited broiler.astrometry.net/~dstn/4200 and 403 permission error. Wide Field 4100 files are accessible OK.

rkaczorek, 2021-04-04:
Why don't you install index files from within Ekos. It's easy and it works. Go to Ekos Align tab, click Options in lowe right corner, choose Index Files on the left. Make sure that Index Files Location drop-down menu points to your /home/astroberry directory and tick recommended index files. These are recommended based on your actual setup and CCD FOV.
Anyway, index files are not distributed by Astroberry and are available from public apt repositories.

len5771, 2021-04-04:
Hi Radek,
That is the point. The left hand group of index files selectors appear to be disabled. I can click away and get nothing except the window jerking to the right as though there is an error. I can add and remove the right hand group at will.
Even after the lengthy download of the 2mass files they don't appear to be accessible.
I have done this on previous installs without a problem so know what normally happens.

rkaczorek, 2021-04-04:
If click boxes are disabled, it means that you do not select writable directory in Index Files Location drop-down menu.

len5771, 2021-04-05:
Hi Radek,
/home/astroberry/.local/share/kstars/astrometry when selected shows in the dark box under that it is enabled. I can add and remove the index-4119 to 4107 fits files at will. While that folder is selected I can't add the 2mass files. If I select the /usr/share/astrmometry folder the dark box under shows Downloads Disabled directory permissions issue. I can't add or remove files as all of the tick boxes are disabled. 
There is as well as the Tycho2 files a astrometry.cfg file in the former but only the direct downloaded 2mass files in the latter. The 2mass files are inaccessible to the solver. All from a fresh install and update/upgrade. Then I downloaded the 2mass with sudo apt install astrometry-data-2mass.

len5771, 2021-04-05:
Checking on my other install and I have used the /home/astroberry/.local/share/kstars/astrometry on that. I have already got the 2mass 4208 to 4213 and 4207-* and 4206-* files and the full set of Tycho2 (not that I need them, just that I can install and remove them at will). I cannot add any other 2mass files. The download progress bar flashes and closes. I assume that the target location is not accessible? I also have the permissions issue showing for the /usr/share/astrometry folder.

rkaczorek, 2021-04-05:
Let's make it clear ;-)

**First, location of configuration and index files.**
Index files location is kept in the following files:
- /home/astroberry/.local/share/kstars/astrometry/astrometry.cfg (used by Kstars/Ekos)
- /etc/astrometry/astrometry.cfg (used by local astrometry.net engine)

Let's focus on the first one, but all of the following apply to the second one as well.

You can expect the following default entries in astrometry.cfg:
```
add_path /usr/share/astrometry
add_path /home/astroberry/.local/share/kstars/astrometry

```
You can add (using Ekos options or by editing the file) any location you like to this file eg.
`add_path /mnt/data/astrometry
`
A solver will expect index files in these directories. Insect each of them and note index files.

**Second, access permissions**
Running Astroberry you use astroberry user, which is unprivileged account. And this is the way it should be.
Anything in /home/astroberry can be accessed by astroberry user as all the files in this directory are owned by astroberry user.
This applies to /home/astroberry/.local/share/kstars/astrometry and does not apply to /usr/share/astrometry
You can easily check the permissions on these directories and configuration files eg.
```
ls -ald .local/share/kstars/astrometry
drwxrwxr-x   2 astroberry astroberry       4096 Nov  2  2019 astrometry
ls -ald /usr/share/astrometry
drwxr-xr-x 2 root root 4096 Feb 21  2020 /usr/share/astrometry

```
The first one is owned by astroberry user and group and access rights are set to rwxrwxr-x i.e. owner can (r)read, (w)rite and e(x)ecute the directory, group can also rwx but others can only r and x as there is no w permission set to w (dashed).
The second one is owned by root user and group and access rights are set to rwxr-xr-x. As you can see only owner can rwx, group can r-x and other can r-x. This means that you cannot write to this directory as astroberry user.

What can you do about it? It is not good idea to run KStars as root so you can change ownership of the folder or permissions. However this is not the best approach as these are set for purpose. If you want to go this way you can run `sudo chown astroberry /usr/share/astrometry` or  `sudo chmod 777 /usr/share/astrometry` respectively. Note that if you want to change ownership of all the content of the directory you need to run `sudo chown astroberry -R /usr/share/astrometry` for recursive action. I do not recommend going this way because it is just a workaround.

Instead you can copy all index files to a folder you own eg. /home/astroberry/.local/share/kstars/astrometry and keep going from there. After copying these files you can safely remove source files. No harm to the system whatsoever, except you need to make sure that /etc/astrometry/astrometry.cfg does not use this source anymore.

I hope it is helpful. If it doesn't work nothing will ;-)

parralix, 2021-04-05:
i have same issue.  i can download tycho2 indexes but i cant download 2mass ones...   i see a quick download icon flash and nothing else

len5771, 2021-04-06:
https://groups.google.com/g/astrometry/c/GllAernE5Ic

len5771, 2021-04-08:
The Astrometry files are accessible again. Maybe a user definable astrometry file URL would help in these cases.

Bowserambo, 2025-02-27:
Same issue on latest update/upgrade on astroberry rpi4. Made dir writable as via your suggestion, but download still doesn't work for 2mass. 2mass files are in folder /usr/share/astrometry but not recognized/visible/downloadable in Kstars/Ekos. Tycho2 works fine. Let me know if you need more specifics pls?

avichalk, 2025-06-07:
Having the same issue on a new astroberry rpi4 install. Tycho catalog files download just fine but 2mass fails each time. @Bowserambo as a temporary workaround you can download the files manually, the following syntax in the terminal worked for me.

1) Note the required files as noted in the Ekos GUI. E.g. in my case index-4207-*.fits was required.
2) Head to https://data.astrometry.net/4200/ and note the ranges of your required files. E.g. for my case we have index-4207-00.fits to index-4207-11.fits.
3) 
```
cd /home/astroberry/.local/share/kstars/astrometry
wget https://data.astrometry.net/4200/index-[number]-{[min]..[max]}.fits
```

E.g. `wget https://data.astrometry.net/4200/index-4207-{00..11}.fits`. This syntax can also be used to iterate over indices, e.g. `index-{4205..4207}-{00..11}.fits` but I find it better to do it manually since some indices only contain one file (like index-4208.fits). On reloading the configuration menu once the download is completed, the index files are marked as downloaded.