rkaczorek/astroberry-server #52: Disappearance of the desktop

THX59, 2019-12-13:
Hi,
I tried your image on RPI3 for few weeks without any problems.
Now I'm trying it on RPI4 but my all the buttons on the desktop are gone after set the resolution.
Only the right button of the mouse runs. The desktop is empty.
Do you have an idea ?
Regards

rkaczorek, 2019-12-13:
Are you sure you are on the right workspace? You can switch between workspaces using switcher on te taskbar

THX59, 2019-12-14:
Hi,
The problem is that there are no taskbar...
[https://drive.google.com/file/d/1fRwZF1n6J2pN37EftwvAIrozV45FFBF9/view?usp=sharing](url)
Any ideas before rewrite all the SD card ?

rkaczorek, 2019-12-15:
Looks like key desktop components do not start at all. You can try to run things manually.
1. Connect to your desktop as you do
2. Login to your astroberry with SSH (parallely)
3. Run in terminal:
`export DISPLAY=:0
lxpanel --profile LXDE-astroberry
`
If this will change the desktop, there must be some failure in starting desktop components.
Before reflashing make copy and remove this directory /home/astroberry/.config/lxpanel
This is where desktop customizations are saved. Maybe you introduced a change that prevents desktop environment start properly. Just guessing

THX59, 2019-12-15:
Hi,
I have done what you have suggested.

astroberry@astroberry:~ $ export DISPLAY=:0
astroberry@astroberry:~ $ lxpanel --profile LXDE-astroberry
There is already an instance of LXPanel.  Now to exit

The desktop is still empty so I will reflash the SD card.
I precise that I didnt have this problem with the RPI3 before.
Regards

closed 2019-12-15 by THX59