rkaczorek/astroberry-server #53: Connect to home wifi

troske, 2020-02-02:
I am trying to follow the instructions listed for connecting to home WiFi. I am connected to the hotspot and controlling through my iPad. I want to switch over to my home WiFi so that the Pi4 Astroberry can be on the internet for downloading updates. When I click on the WiFi icon, it shows astroberry, but no other available WiFi networks. My phone and iPad can see lots of network IDs from my house and the neighbors, but I don’t know how to make them show up in the Pi so that I can choose an available network and connect. 

I have been searching online, but have only found pages of command lines and configuration files that need to be created by a computer programmer, way too complicated for me. I must be missing something because connecting to an existing WiFi network should only take a click or two.

rkaczorek, 2020-02-02:
While Astroberry Hotspot is active no other wireless network can be used. You are either connected to your home network or Astroberry Hotspot  (if home network is not available).

Wireless connection is predefined for you. Just edit it and change network name and password.

- Right-click wireless icon on the taskbar
- Select Edit connections
- Double-click Wireless connection
- Enter your network name in SSID field
- Go to Wi-Fi Security tab
- Enter your network password in Password field
- Reboot

cstachon, 2020-02-04:
Just to add to Radek's comments  - The RPI Hotspot  is used by most users only when "in the field" i.e. away from home.      In this "in the field" scenario the user will connect their Ipad/Tablets etc to the RPI hotspot which then gives them access to Astroberry functions. No Internet is available in this scenario. Plus the RPI Hotspot will be automatically be available without user intervention. 

 At home most users just connect the RPI Wifi (acting as a normal client just like your Ipad etc) to their home Wifi network or just insert a wired Ethernet connection. Connected this way (Wifi or wired)  the RPI can access the Internet (for updates etc) and any device on your Home network can connect to the RPI. 

There are no files to edit to achieve either the above modes of working - just follow Radek's Instructions to add your home WiFi network - thats it no computer programmers involved :-)

Radek please delete this if you think its not appropriate :-)

Manu4810, 2020-04-26:
Hi, just joining Astroberry. Looks fantastic, will soon replace my laptop. I have a issue however regrading the Wifi configuration. I understand that it is either hot spot, or "classical" client wifi, but not both at the same time.  managed to switch from WAP to "classical" wifi client at home and connect to the internet, but now I'm not able to switch back to the Wifi AP (rebooting does not change the wifi config), and am stuck in the client mode. Is it possible to re-activate the wifi-AP through the GUI or do you need command line ? Thanks !

rkaczorek, 2020-04-27:
wifi-AP will work if your wifi AP is not available, so either switch if off to test it or change it's name or password in Astroberry so it can't login

Manu4810, 2020-05-01:
Works fin, thanks ! There seem to be also the possibility to play with the priority level within advanced network configuration, assigning a lower value to the router Wifi vs the RPI Wifi-AP.

frankdurham, 2020-08-02:
Okay, I asked this in another forum with No answer.. 
But  I'm trying to figure out how to configure Astroberry so that I can get it to UPDATE ..  I see people saying they can access internet and Astroberry server. .but everything up here says NO.. it's One or the other? But that doesn't make sense? how else could you get updates? 
So I'm asking in here.. How to configure Astroberry to both access the server from my laptop. and have access to the internet at the same time? 
Can I just mess with the WPA file in Buster to access my internet info? so when it boots up it will hopefully boot up still correctly??  
 I see a msg with a link on the bootup of the server that mentions something about adding it. but when I click the link. obviously it goes nowhere cuz I don't have internet access when on the server?

rkaczorek, 2020-08-02:
No need to mess with anything. Connect your astroberry to your home network. Next, instead of connecting to astroberry hot spot, connect to it within your home network (you can check it's ip on your home network AP/router or access it using astroberry.local). That's all

frankdurham, 2020-08-03:
Thanks for the quick reply Radek,
That's what I thought.. but someone told me, if I did that Astroberry would boot to Astroberry and just stop at my network.
Im thinking he missunderstood my question lol..
Hope all is well with you and yours during this Covid crap! 
Frank

amartinwest, 2021-01-21:
Apologies for piggy backing this thread, just installed astroberry but I didnt get a wifi icon in the top bar, until I realised that the astroberry desktop was a zillion pixels wide and I eventually found it by repeatedly expanding the window and dragging it left until the end of the top bar appeared where I then found the said wifi one.

Just have to figure how to scale it, firstly I use the browser view zoom out, then I found a scale function in the settings on the left side, local mode seemed to work.

Im running on 15"  MacBook Pro

Impressive implementation btw. Thanks

frankdurham, 2021-01-23:
Hi Martin.
Sorry your having issues with resolutions.
Changing the Raspberry Pi's screen resolution   
   - To change the resolution, run the command 
   - sudo raspi-config 
   -  navigate to Advanced Options > Resolution, and choose an option.
   - If this menu is not available, or you want more control, specify settings in the /boot/config.txt file
Sent from Yahoo Mail on Android

amartinwest, 2021-01-23:
Not a problem, it was easy to resolve, once I twigged what the problem was :-),  I have some experience with linux and have been programming for over 50 years.

I am most impressed with the astroberry system. 

Thank you

regards Martin West
skype:amartinwest
mob:07525424513

frankdurham, 2021-01-24:
That would be Radek’s doing.. 
He did a great job. !! 
I’m not even sure how I got on this thread? LOL 

Frank

From: Martin West
Sent: Saturday, January 23, 2021 3:06 PM
To: rkaczorek/astroberry-server
Cc: frankdurham; Comment
Subject: Re: [rkaczorek/astroberry-server] Connect to home wifi (#53)

Not a problem, it was easy to resolve, once I twigged what the problem was :-), I have some experience with linux and have been programming for over 50 years.

I am most impressed with the astroberry system. 

Thank you

regards Martin West
skype:amartinwest
mob:07525424513

PRToxastro, 2021-07-31:
Wondering if there are any more resolutions to try to get the home network attached to the RPi? I have been trying many things throughout the web and on this discussion and have had no resolution. After following the instructions and then rebooting it does not connect to my home network and I still have to login through the astroberry hotspot. I have tried everything in changing the Wireless Connection to my home settings but to no avail. Sure would be nice to be able to access RPi through my home network though.

mezdog, 2023-10-07:
Can a run a debugging test on my iPad Pro? My neighbor has been using/taking over my iPad Pro. I’m a novice at this sort of stuff so I can surely use help. Thank you , Mezdog