rkaczorek/astroberry-server #54: QHY Cameras not working

sonny486, 2020-02-17:
QHY cameras were working before the latest update.  Now QHY 163M does not work now.

:-(

sonny486, 2020-02-18:
From Indilib.org:

It is related to last update of libqhy... Please downgrade the library by running:
wget https://www.astroberry.io/repo/pool/main/libq/libqhy/libqhy_6.0.5-2_armhf.deb
sudo dpkg -i libqhy_6.0.5-2_armhf.deb

sonny486, 2020-02-18:
BTW Thanks for your hard work on this!   I love this system, made my imaging so much easier!  Thanks again.

rkaczorek, 2020-02-18:
Please downgrade the library by running:
`wget https://www.astroberry.io/repo/pool/main/libq/libqhy/libqhy_6.0.5-2_armhf.deb
sudo dpkg -i libqhy_6.0.5-2_armhf.deb`

sonny486, 2020-02-18:
Thanks, I have done that.  I didn't realize it was you who responded in Indilig.org.  Thanks.  It works great now!   I do appreciate your work.

closed 2020-02-18 by sonny486