rkaczorek/astroberry-server #57: GPS simulator not using new location data

basilr, 2020-02-26:
After a fresh install of v2.0.1 on a RPi 3 I updated the lat and long as described in the FAQ's. This worked fine. After running an update, sudo apt update, sudo apt upgrade, astroberry defaults back to the default lat and long (see screen shot). Also, there is a flashing "waiting" next to Fix. If I change the lat and long using Preferences->Geographic Location, the change doesn't work for the GPS Info and the Celestial Info. Perhaps a permissions issue?

[image]

Many thanks for a fantastic piece of work!

rkaczorek, 2020-02-26:
Does it refresh after reboot or after restarting virtualgps?
`sudo systemctl restart virtualgps.service`

basilr, 2020-02-26:
Nope. Also, after reading issue 55, I checked for /tmp/vgps, which does not exist anywhere.

Any log files I can send?

rkaczorek, 2020-02-26:
Fixed in virtualgps_1.0.1-1

closed 2020-02-26 by rkaczorek

basilr, 2020-02-28:
Hi, Sorry to bring this up again, but it seems to still be doing this.
I did a fresh install from the image, v2.0.1. Then ran sudo update, sudo upgrade. Nothing else, no other changes. No change to the lat and long using Preferences->Geo Location. Then a reboot.

The GPS Info screen looks like the above screenshot, no lat, long, UTC time displayed, just flashing "waiting" next to Fix. Changing the location in Preferences-> Geo Location and rebooting still shows the above screen. Tried clearing my browser cache, no difference.

Not a big issue for me, I just really like the Polar Finder feature and would like to use it.
Thanks