rkaczorek/astroberry-server #59: CR3 files and libraw

karolbe, 2020-03-06:
Are there any plans to add support for Canon CR3 RAW format? Support is already in libraw library but in order to add it kstars/inidi/ekos has to be recompiled with the new library version.

rkaczorek, 2020-03-26:
I think it requires recompiling core indi libs. If so, it can be added in the next full release of libindi and not earlier.

karolbe, 2020-03-27:
I have compiled indi and gphoto (not sure if this one is required but I saw dependency and decided to recompile as well) from sources, this allows me to see images captured by EOS R when format is set to FITS. Not ideal but this is enough to use plate solving etc.

To view CR3 images in KStars it has to be recompiled as well.

rkaczorek, 2020-04-13:
New versions has been just released. Can you do some testing on EOS R?

karolbe, 2020-04-14:
I have installed 2.0.1 and then I did apt update && apt upgrade and looks like it does not work, or perhaps I did something wrong. I tried with FITS and Native format, both failed. 

Below is the debug log. Let me know how can I help you troubleshoot it.

2020-04-14T16:15:01: [INFO] Starting 1 seconds exposure. 
2020-04-14T16:15:01: [ERROR] Exposure failed to parse raw image. 
2020-04-14T16:15:01: [ERROR] Cannot open /tmp/indi_KL1iju: Unsupported file format or not RAW file 
2020-04-14T16:14:56: [INFO] Starting 1 seconds exposure. 
2020-04-14T16:14:56: [ERROR] Exposure failed to parse raw image. 
2020-04-14T16:14:56: [ERROR] Cannot open /tmp/indi_5WCMIo: Unsupported file format or not RAW file 
2020-04-14T16:14:52: [INFO] Starting 1 seconds exposure. 
2020-04-14T16:14:33: [INFO] Starting 1 seconds exposure. 
2020-04-14T16:14:08: [ERROR] Invalid frame requested (0,0) (0 x 0) 
2020-04-14T16:14:08: [INFO] Device configuration saved. 
2020-04-14T16:14:08: [INFO] Saving device configuration... 
2020-04-14T16:12:33: [INFO] World Coordinate System is enabled. 
2020-04-14T16:12:33: [INFO] No previous configuration found. To save driver configuration, click Save Configuration in Options tab 
2020-04-14T16:12:33: [INFO] Detected Canon.Inc Model Canon EOS R. 
2020-04-14T16:12:33: [INFO] Canon DSLR EOS R is online. 

[2020-04-14T18:28:34.600 CEST INFO ][           org.kde.kstars.ekos] - "Enabling debug logging for Guide Simulator..."
[2020-04-14T18:28:34.603 CEST INFO ][           org.kde.kstars.ekos] - "Enabling debug logging for Canon DSLR EOS R..."
[2020-04-14T18:28:34.610 CEST INFO ][           org.kde.kstars.indi] - Telescope Simulator :  "[INFO] Saving device configuration... "
[2020-04-14T18:28:34.612 CEST INFO ][           org.kde.kstars.indi] - Telescope Simulator :  "[INFO] Device configuration saved. "
[2020-04-14T18:28:34.613 CEST INFO ][           org.kde.kstars.indi] - Guide Simulator :  "[INFO] Debug is enabled. "
[2020-04-14T18:28:34.613 CEST DEBG ][           org.kde.kstars.indi] - < Guide Simulator >: < DEBUG_LEVEL >
[2020-04-14T18:28:34.617 CEST DEBG ][           org.kde.kstars.indi] - < Guide Simulator >: < LOGGING_LEVEL >
[2020-04-14T18:28:34.621 CEST DEBG ][           org.kde.kstars.indi] - < Guide Simulator >: < LOG_OUTPUT >
[2020-04-14T18:28:34.626 CEST INFO ][           org.kde.kstars.indi] - Guide Simulator :  "[INFO] Saving device configuration... "
[2020-04-14T18:28:34.626 CEST INFO ][           org.kde.kstars.indi] - Guide Simulator :  "[INFO] Device configuration saved. "
[2020-04-14T18:28:34.628 CEST INFO ][           org.kde.kstars.indi] - Canon DSLR EOS R :  "[INFO] Debug is enabled. "
[2020-04-14T18:28:34.628 CEST DEBG ][           org.kde.kstars.indi] - < Canon DSLR EOS R >: < DEBUG_LEVEL >
[2020-04-14T18:28:34.633 CEST DEBG ][           org.kde.kstars.indi] - < Canon DSLR EOS R >: < LOGGING_LEVEL >
[2020-04-14T18:28:34.638 CEST DEBG ][           org.kde.kstars.indi] - < Canon DSLR EOS R >: < LOG_OUTPUT >
[2020-04-14T18:28:34.705 CEST INFO ][           org.kde.kstars.indi] - Canon DSLR EOS R :  "[INFO] Saving device configuration... "
[2020-04-14T18:28:34.706 CEST DEBG ][           org.kde.kstars.indi] - Guide Simulator : "[DEBUG] Toggle Debug Level -- Driver Debug "
[2020-04-14T18:28:34.708 CEST INFO ][           org.kde.kstars.indi] - Canon DSLR EOS R :  "[INFO] Device configuration saved. "
[2020-04-14T18:28:34.708 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Toggle Debug Level -- Alignment Subsystem "
[2020-04-14T18:28:44.334 CEST DEBG ][   org.kde.kstars.ekos.capture] - Preparing capture job "/home/astroberry/Light" for execution.
[2020-04-14T18:28:44.337 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1.000-second  image..."
[2020-04-14T18:28:44.396 CEST INFO ][           org.kde.kstars.indi] - Canon DSLR EOS R :  "[INFO] Starting 1 seconds exposure. "
[2020-04-14T18:28:44.396 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Starting exposure (exptime: 1 secs, mirror lock: 0) "
[2020-04-14T18:28:44.397 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Mutex locked "
[2020-04-14T18:28:44.397 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Setting radio/menu widget imageformat: 21 (RAW) "
[2020-04-14T18:28:44.398 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Setting new configuration OK. "
[2020-04-14T18:28:44.399 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Using internal bulb widget:eosremoterelease "
[2020-04-14T18:28:44.399 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Setting radio/menu widget eosremoterelease: 2 (Press Full) "
[2020-04-14T18:28:44.535 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Setting new configuration OK. "
[2020-04-14T18:28:44.536 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Exposure started. "
[2020-04-14T18:28:44.536 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Time left: 999 ms "
[2020-04-14T18:28:45.536 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Time left: -1 ms "
[2020-04-14T18:28:45.536 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Closing internal shutter. "
[2020-04-14T18:28:45.537 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Using widget:eosremoterelease "
[... cut ...]
[2020-04-14T18:29:05.714 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Event timed out #7, retrying... "
[2020-04-14T18:29:06.718 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Event timed out #8, retrying... "
[2020-04-14T18:29:07.724 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Event timed out #9, retrying... "
[2020-04-14T18:29:08.728 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Event timed out #10, retrying... "

rkaczorek, 2020-04-14:
> [2020-04-14T18:28:49.081 CEST DEBG ][ org.kde.kstars.indi] - Canon DSLR EOS R : "[DEBUG] Cannot decode (Unsupported file format or not RAW file) "

Please install libraw and try again. I think this is missing dependency
`sudo apt install libraw-bin`

karolbe, 2020-04-14:
Well, libraw-bin is version ibraw-bin armhf 0.19.2-2 which as far as I know does not support CR3.

rkaczorek, 2020-04-14:
Oh, so it is Raspbian Buster issue than... I'm not packaging this binary.

siberx, 2021-04-28:
For anybody still encountering this issue, I recently ran into the same problem with a Canon EOS M200 and the latest Astroberry v2.0.3 build. I could connect to the camera fine and control it, but RAW downloads (in Native or FITS format) always failed. As noted above, this is because the included LibRaw is 0.19.2 and initial (uncompressed only, which is all we need) CR3 support isn't available until LibRaw 0.20.x.

After a bunch of testing, I determined that the following sequence needs to happen from a fresh Astroberry image to get CR3 support working correctly:
* Update the newly deployed image to the latest apt packages
* Build the current stable v0.20.2 LibRaw library from source and install it
* Rebuild the indi core library from source and install it (you don't technically need this updated to fix the problem, but the indi-3rdparty drivers won't build correctly unless you update this first)
* Rebuild the indi-gphoto driver from source (this has a LibRaw dependency and definitely must be rebuilt so the driver itself doesn't throw errors) and install it
* Rebuild Kstars from source (this also has a LibRaw dependency and must be rebuilt for FITS viewer to work) and install it

After the above sequence, I was able to use Kstars/Ekos with the indi-gphoto driver for Canon cameras and get downloads working in both FITS and Native/CR3 modes. This same fix should apply to any Canon cameras that use DIGIC 8 (or newer?) processors, which is basically anything released from 2018/2019 onwards. The R/RP/R5/R6/Ra, M50, M50 MkII, 90D, 850D Rebel SL3/250D, M6 MkII and M200 are all the ones I expect are affected - I was pretty surprised to discover that Astroberry is effectively unusable with all these new-ish Canon cameras out of the box, considering how common Canon cameras are in the astrophotography world. I specifically moved from an old Sony NEX-5R to a modern Canon camera to get proper software support, but I guess I went straight from "too old" to "too new" 😉

Unless LibRaw gets repackaged directly into the Astroberry repositories, a proper fix for this probably won't come until Debian Bullseye goes stable (since the v0.20.x LibRaw libraries don't show up in Buster, only Bullseye testing), then Raspberry Pi OS releases their downstream distro on Bullseye, then Astroberry gets rebuilt off of a new Bullseye base. Since Bullseye has only recently gone into hard freeze in early March, it will likely still be a while before this all rolls through.

For reference, my commented command sequence is included below in case anybody finds it helpful to assist in getting this CR3 support working, as it will likely be months at least before it's included natively in Astroberry.

```
sudo apt update
sudo apt -y upgrade
#=============================================
# LibRaw
sudo apt remove libraw-bin
mkdir -p ~/Projects/libraw
cd ~/Projects/libraw
sudo apt -y install autotools-dev automake libtool
wget https://www.libraw.org/data/LibRaw-0.20.2.tar.gz
tar -zxf LibRaw-0.20.2.tar.gz
cd LibRaw-0.20.2
autoreconf --install
./configure
make
sudo make install
sudo ldconfig

#=============================================
# Core indi library
sudo apt -y install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libtiff-dev libfftw3-dev libftdi-dev libgps-dev libdc1394-22-dev libgphoto2-dev libboost-dev libboost-regex-dev librtlsdr-dev liblimesuite-dev libftdi1-dev libavcodec-dev libavdevice-dev
# Removed libraw-dev from the above as we're providing a newer one from source
mkdir -p ~/Projects
cd ~/Projects
git clone --depth 1 https://github.com/indilib/indi.git
mkdir -p ~/Projects/build/indi-core
cd ~/Projects/build/indi-core
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi
make -j4
sudo make install

#=============================================
# indi-gphoto
mkdir -p ~/Projects
cd ~/Projects
git clone --depth=1 https://github.com/indilib/indi-3rdparty
mkdir -p ~/Projects/build/indi-gphoto
cd ~/Projects/build/indi-gphoto
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-gphoto
make -j4
sudo make install
sudo cp indi_gphoto.xml /usr/share/indi

#=============================================
# kstars (building stellarsolver from source to skip ppa, omitting libraw to use our compiled version)
sudo apt -y install build-essential cmake git libeigen3-dev libcfitsio-dev zlib1g-dev libindi-dev extra-cmake-modules libkf5plotting-dev libqt5svg5-dev libkf5xmlgui-dev libkf5kio-dev kinit-dev libkf5newstuff-dev kdoctools-dev libkf5notifications-dev qtdeclarative5-dev libkf5crash-dev gettext libnova-dev libgsl-dev libkf5notifyconfig-dev wcslib-dev libqt5websockets5-dev xplanet xplanet-images qt5keychain-dev libsecret-1-dev breeze-icon-theme
mkdir -p ~/Projects
cd ~/Projects
git clone --depth=1 https://github.com/rlancaste/stellarsolver.git
mkdir ~/Projects/stellarsolver-build
cd ~/Projects/stellarsolver-build
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTER=ON ~/Projects/stellarsolver/
make -j $(expr $(nproc) + 2)
sudo make install

# Swap file size change (since kstars fails to build with 4GB ram on a pi4 with the 100MB swap default):
sudo dphys-swapfile swapoff
sudo nano /etc/dphys-swapfile
	# Edit CONF_SWAPSIZE to 2048
sudo dphys-swapfile setup
sudo dphys-swapfile swapon

mkdir -p ~/Projects
cd ~/Projects
git clone --depth=1 https://invent.kde.org/education/kstars.git
mkdir -p ~/Projects/kstars-build
cd ~/Projects/kstars-build
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=RelWithDebInfo ~/Projects/kstars
make -j8
sudo make install
```

kevinmilitello, 2021-05-06:
@siberx thanks for the detailed information. I'm currently stuck and will try this out. Does this allow the new polar alignment tool in KStars 3.5.2 and plate solving to work? I've had issues with these because presumably those require FITS which fails when the Canon captures .CR3.

siberx, 2021-05-07:
Assuming your attempts were failing due to an inability to download and parse CR3 files and not for some other reason, this should resolve that. I haven't had a chance to do any imaging yet since I figured out the recompile process so I haven't attempted any plate solving or alignment through KStars directly, but I can definitely now capture previews or exposures in either FITS or Native format with my M200 and get them to view/render correctly in the Kstars FITS viewer, so I expect the same will apply to the other tools.

Note that this also means you'll be using the latest master branch builds for the tools you recompiled, which means your INDI libraries and Kstars might be running newer builds than what's available in the Astroberry repositories. I assume that what those projects make available in their default branches is considered fairly stable, but just a heads up in case you notice other differences before/after running the above commands.

Also note that if there are _other_ tools you use in your stack that depend on LibRaw then they might still use/refer to the old version and thus won't work just with the above command sequence. It's possible CCDciel, oaCapture, FireCapture, ASTAP, etc... might use the library, but I don't use those in my workflow so I didn't bother investigating them any further. If you do use these, feel free to test them and report back!

kmilitello, 2021-05-08:
@siberx It looks like this will work. I was able to take images.... the problem now is of course the clouds to verify.

d33psky, 2021-05-09:
I may have an alternative solution until astroberry / raspbian upgrades to a newer OS version.
I've recompiled the libraw package and tricked the linker (the ugly part) and made .deb installation packages.
But I cannot test myself due to my camera being too old and it having a broken USB socket.
Zip file and installation instructions are here -> https://indilib.org/forum/astroberry/7875-astroberry-and-cr3-canon-250d.html#71128

Work done:
Based on source package at https://packages.debian.org/source/bullseye/libraw
First let the current version build, to test and install all build dependencies
```
apt-get source libraw
apt-get build-dep libraw
apt-get --build source libraw
```
This produces: 
```this produces :
drwxr-xr-x 17 root root   4096 May  9 22:04 libraw-0.19.2
-rw-r--r--  1 root root  70248 May  9 22:04 libraw-bin_0.19.2-2_armhf.deb
-rw-r--r--  1 root root 171664 May  9 22:04 libraw-bin-dbgsym_0.19.2-2_armhf.deb
-rw-r--r--  1 root root 282136 May  9 22:04 libraw19_0.19.2-2_armhf.deb
-rw-r--r--  1 root root 288304 May  9 22:04 libraw-dev_0.19.2-2_armhf.deb
-rw-r--r--  1 root root  81840 May  9 22:04 libraw-doc_0.19.2-2_all.deb
-rw-r--r--  1 root root 971120 May  9 22:04 libraw19-dbgsym_0.19.2-2_armhf.deb
-rw-r--r--  1 root root   6236 May  9 22:04 libraw_0.19.2-2_armhf.buildinfo
-rw-r--r--  1 root root   2705 May  9 22:04 libraw_0.19.2-2_armhf.changes
```

Next build the bullseye version of 0.20:
```
wget http://deb.debian.org/debian/pool/main/libr/libraw/libraw_0.20.2-1.dsc
wget http://deb.debian.org/debian/pool/main/libr/libraw/libraw_0.20.2.orig.tar.gz
wget http://deb.debian.org/debian/pool/main/libr/libraw/libraw_0.20.2-1.debian.tar.xz
dpkg-source -x libraw_0.20.2-1.dsc
cd libraw-0.20.2/
debuild
```

This produces:
```
this produces :
-rw-r--r--  1 astroberry astroberry   88716 May  9 22:24 libraw-bin_0.20.2-1_armhf.deb
-rw-r--r--  1 astroberry astroberry  273220 May  9 22:24 libraw-bin-dbgsym_0.20.2-1_armhf.deb
-rw-r--r--  1 astroberry astroberry  341788 May  9 22:24 libraw20_0.20.2-1_armhf.deb
-rw-r--r--  1 astroberry astroberry  368368 May  9 22:24 libraw-dev_0.20.2-1_armhf.deb
-rw-r--r--  1 astroberry astroberry   90524 May  9 22:24 libraw-doc_0.20.2-1_all.deb
-rw-r--r--  1 astroberry astroberry 1269404 May  9 22:24 libraw20-dbgsym_0.20.2-1_armhf.deb
-rw-r--r--  1 astroberry astroberry    6463 May  9 22:24 libraw_0.20.2-1_armhf.buildinfo
-rw-r--r--  1 astroberry astroberry    3362 May  9 22:24 libraw_0.20.2-1_armhf.changes
-rw-r--r--  1 astroberry astroberry  226929 May  9 22:24 libraw_0.20.2-1_armhf.build
```

Installation and symlink mangling in the indilib.org post.

-- Hans

tstevens83, 2021-05-17:
I confirmed the steps provided by siberx... thank you !!  Now I can work on completing my astroberry setup with my equipment.  Looking forward to clear skies !!

Tim

jscheidtmann, 2021-05-18:
Libraw 0.20.2 is in testing (bullseye). Could you try upgrading to testing and see if it solves the problem?

siberx, 2021-05-18:
@jscheidtmann I considered this approach first since it involves less actual steps to do, but I didn't find a lot of good information on cleanly upgrading a Pi to bullseye and whether or not the Astroberry repositories would support that (is there a bullseye distro in the astroberry repo currently?) or whether there would be additional dependency conflicts. In other words, it seemed a lot more disruptive than just recompiling the specific packages I knew needed it (since the whole OS would be upgraded), so I did not attempt the full upgrade.

Anybody else who is bolder than me can certainly give it a shot though, but I don't want to mess with my current solution too much as I'm just glad it's working for now. Even if upgrading to bullseye works, it still means having to mess with the stock Astroberry image quite a bit until it's pulled into stable so the re-compile solution works for me for now.

jscheidtmann, 2021-05-19:
@siberx The upgrade involves switching from buster to bullseye in /etc/apt/sources.list(.d/*). This is a per repository thing, so you can leave alone astroberry's. Most of the Astroberry packages do configuration, so there are not many dependencies on libraries. YMMV, though. Maybe you've got a spare SDCard?

damianjableka, 2021-06-03:
@siberx solutions indeed works very well, my Canon 90D start works perfectly, but Eqmod Mount and SVBONY sv305 drivers crashed at INDI server start, should I compile something more, or it is not possible to get everything to work?

siberx, 2021-06-03:
@damianjableka I don't have any SVBONY devices so I'm not sure what its dependencies are. I see an `indi-sv305` driver in the `indi-3rdparty` repo, so if that's what you're using and it has a LibRaw dependency then you might need to recompile it as well. You can follow the same pattern as shown in my command list under the `indi-gphoto` section and just substitute `indi-sv305` in anywhere it says `indi-gphoto` as you'd basically just be compiling a different sub-section of the indi-3rdparty set in this case.

 I'm surprised you mention crashing with EQMOD; I have been using that without issue on my installation without having to do anything special, and in general you wouldn't think that a mount driver would need to interact with a RAW image library/dependency at all. In any case, if you find it still crashes even with the sv305 driver recompiled, then you can give the same above approach a shot with `indi-eqmod` since it's also present in the indi-3rdparty repo.

damianjableka, 2021-06-04:
@siberx Thank you for your suggestion, and fast answer! It really helped, and reduce my stress. After recompilation of both SV305 and EQMod it works now.  As I work remotely I can see that everything run, but unfortunately all devices are physically disconnected, so I will check everything  when I attach the.

It is really weird that EQMod crashed, what connet it to LibRaw? Fortunately recompilation helped.
[image]

simra, 2022-01-16:
I can successfully build and run @siberx's approach, but I only receive single-channel images from my Canon M50 camera- at least the previewer shows only one channel, and the saved FITS file is single-channel.  I've tried a few combinations of settings in INDI, as well as both the canon ccd and gphoto drivers.  The only time I get color images is when I launch the live video preview.  Maybe this is a configuration issue? I'm new to kstars and don't see any options to toggle RGB vs grayscale.

[image]

ppoinha, 2022-02-06:
Hi All,

When trying to compile the final step of the object "kstars" I am getting the below error.
Any help, or should I just wait for an update that might work.

I followed the guide step by step.

[image]

Thanks,
Paulo Poinha

siberx, 2022-02-07:
@ppoinha I haven't had to do this build recently (still using my install from when I wrote it up) but it's possible that something has changed in kstars or one of its dependencies recently that breaks the compilation process. This should (hopefully) get fixed quickly because I assume others will notice this too, but I'm not 100% sure. 

The guide as documented builds the "latest" version which might be fairly bleeding-edge, so you can try pulling down an older commit or branch for kstars instead to see if you can get it to build, but I haven't dug through their repo to figure out how they handle versioning to have a specific recommendation.

rkaczorek, 2022-02-07:
@ppoinha you are missing [libstellarsolver](https://github.com/rlancaste/stellarsolver)

ppoinha, 2022-02-07:
@rkaczorek I built the stellarsolver as part of the steps and installed it, so that should not be missing. The step just before I build the kstars is the stellarsolver build. Am I missing something here ?

Thanks.

rkaczorek, 2022-02-07:
Make sure that you use proper version of StellarSolver. 1.9 has been recently released while KStars might link to 1.8 (at least it was the case on the 29th of January when I released last updates to Astroberry)

ppoinha, 2022-02-08:
exactly the issue, but works if we get the stellarsolver dev libraries. glad it is solved now.

Akinetopsia1, 2022-08-25:
i've followed @siberx 's instructions which worked fine until attempting to build kstars... which now relies on QT 5.12.8+ which is unavailable for the astroberry debian version and needs to be compiled from source, which takes 5-6 hours and really doesn't go smoothly (had to recompile 3 times)

3 days later i finally succeed, but now...

```
CMake Warning at /usr/local/share/ECM/find-modules/FindKF5.cmake:52 (find_package):
  Could not find a configuration file for package "KF5NotifyConfig" that is
  compatible with requested version "**5.68.0**".

  The following configuration files were considered but not accepted:

    /usr/lib/arm-linux-gnueabihf/cmake/KF5NotifyConfig/KF5NotifyConfigConfig.cmake, version: **5.54.0**
```

Looks like i'd also have to compile KDE Framework 5 from source.

I give up.

I can generally work my way around a few source compiles but i'm getting way over my head with this. Is there a way to get an older kstar version source that will compile without the need to manually basically recompile most of the OS ? 😫

siberx, 2022-08-25:
@Akinetopsia1 Assuming you're getting that KDE version compatibility warning as part of the KStars build, I would try pulling down one of the older versions of KStars and building that instead. It looks like previous versions of KStars are tracked in their own stable branches, so give `stable-3.5.9` a shot, and if that doesn't work keep decrementing the last digit (`stable-3.5.8`, `stable-3.5.7`...) until you get one that builds correctly and doesn't depend on a newer version of Qt than what's available already.

The only command you should have to change is the git clone for kstars, I think something like the following should work:
`git clone --depth=1 --branch stable-3.5.9 https://invent.kde.org/education/kstars.git`
and if that fails to pull down the branch in question correctly, then the single-branch might do what we need:
``git clone --single-branch --branch stable-3.5.9 https://invent.kde.org/education/kstars.git`

I have not tested this process yet, but feel free to give it a shot if you still have any motivation left and let us know how it goes :)

Akinetopsia1, 2022-08-25:
@siberx ah! thanks! i missed the previous Kstars branches as that was my initial intention to try but i didn't want to have to arbitrarily roll back. Having actual branches makes that easier.

I'm currently restoring the sd backup i made before starting this, as i've since tried jscheidtmann's suggestion to switch to bullseye in the apt sources.list file, did apt update / upgrade a few times, force installed gcc-8-base to clear a conflict, re updated/upgraded and successfully borked everything as it reset the default desktop and hostname, i lost kstars, broke my usb wireless driver and lost the nginx microsite 😅 

i'll give this another go, thanks again

rkaczorek, 2022-08-25:
It compiles properly from stable-3.6.0 branch. You can find it here https://invent.kde.org/education/kstars/-/tree/stable-3.6.0
Anyway it's going to be an issue for following versions as this dependencies are not available on Debian Buster.
Packages for Debian Bullseye for arm64 have been already compiled and are going through testing. It will take just a few days to promote to astroberry repo. This will address the issue as the newest versions of dependencies are available in Bullseye

Akinetopsia1, 2022-08-26:
Success confirmed with kstars **stable-3.6.0** and I have .CR3 files downloading and displaying properly !

I did have to `sudo apt install libev-dev`  in order to compile indi-core, and then i had this message when attempting to compile kstars:

`CMake Error at cmake_modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Libev (missing: LIBEV_LIBRARY LIBEV_INCLUDE_DIR)`

which could be fixed by running

`cd ~/Projects
git clone https://invent.kde.org/frameworks/extra-cmake-modules.git 
cd extra-cmake-modules
mkdir build && cd build    
cmake ..
make && sudo make install`

Thanks a lot @siberx  and @rkaczorek !

Now to thoroughly test everything so that i don't lug a 50 pounds brick out in the field an hour away this weekend 😁

Akinetopsia1, 2022-09-01:
Post-Mortem update: although i thought i thoroughly tested everything at home, out in the field Kstars would completely crash and quit repeatably after the first exposure if i tried to use Ekos exposure automation. Triggering a single exposure would complete, download, and open in the FITS viewer, but it'd crash if i tried to start a sequence.

I ended up manually triggering groups of self-timer sets of 10 exposures directcly from the camera, but that wasn't the only problem i had that night although no fault of astroberry/kstars- just good old Murphy's law. (Broke my tripod leg spider lock, lost my guide scope lock thumbscrew, connection issues and thick fog falling down and wetting everything 🙃 )

I'll try to narrow down the issue once i unpack the kit but right now i'm still angry at it so in the bag it stays.

kchernoff, 2022-09-01:
Well, sounds like you had a very relatable night of astrophotography to me 😂

Keep us posted on what you find with further testing. I believe it's possible to launch KStars/Ekos from a command-line window so that it will output additional logging information, and if not then the log files on disk will probably give additional hints about exactly what is crashing and why.

siberx, 2022-09-01:
Well, sounds like you had a very relatable night of astrophotography to me 😂

Keep us posted on what you find with further testing. I believe it's possible to launch KStars/Ekos from a command-line window so that it will output additional logging information, and if not then the log files on disk will probably give additional hints about exactly what is crashing and why.

Akinetopsia1, 2022-09-05:
so, did a quick test yday which worked, paused the sequence, restarted and kstars crashed two times at the message "preparing" in the progress box. However the log folder did not exist in ~/.local/share/kstars, so today i created it, did another few tests and...  of course everything was working and it was happily snapping away for 3 sequences of varying settings and number of images. 

I then left it alone as i was typing this, went back to try a single exposure, which worked, created a new sequence and hit start, which triggered the camera and then.. crash.

no logs in the log folder. the only thing i can see is that the camera had its screen on while taking the first and second image while it would be off the previous times where it worked.

Maybe a Gphoto2 problem ?

rkaczorek, 2022-09-05:
Enable logging first, using Logs button on Ekos main screen.

Akinetopsia1, 2022-09-06:
enabled ekos versbose and all of the options. no error after the capture, it just closes.

Here's the end of the log -- PHD2 complains because my asi cam is not connected, i'm just running the pi and my EOS M200

```
[2022-09-05T21:38:19.395 EDT DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Exposure complete. "
[2022-09-05T21:38:19.398 EDT DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Downloading image... Name: (IMG_6848.CR3) Folder: (/store_00020001/DCIM/101MEDIA) Delete from SD card? (false) "
[2022-09-05T21:38:20.161 EDT INFO ][     org.kde.kstars.ekos.guide] - "Reconnecting to PHD2 Host: localhost, on port 4400. . ."
[2022-09-05T21:38:20.167 EDT INFO ][     org.kde.kstars.ekos.guide] - "The connection was refused by the peer. Make sure the PHD2 is running, and check that the host name and port settings are correct."
[2022-09-05T21:38:21.034 EDT DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG]  Downloaded 6000x4000 (preview 160x120) "
[2022-09-05T21:38:21.045 EDT DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Capture target is SD Card. "
[2022-09-05T21:38:21.178 EDT DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Uploading file. Ext: CR3, Size: 21034250, sendImage? Yes, saveImage? No "
[2022-09-05T21:38:23.161 EDT INFO ][     org.kde.kstars.ekos.guide] - "Reconnecting to PHD2 Host: localhost, on port 4400. . ."
[2022-09-05T21:38:23.171 EDT INFO ][     org.kde.kstars.ekos.guide] - "The connection was refused by the peer. Make sure the PHD2 is running, and check that the host name and port settings are correct."
[2022-09-05T21:38:25.710 EDT DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-09-06T01:38:25: Driver indi_gphoto_ccd: Received 1 fds"
[2022-09-05T21:38:25.714 EDT DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2022-09-05T21:38:25.714 EDT DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] BLOB transfer took 0.0848041 seconds "
[2022-09-05T21:38:25.714 EDT DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Upload complete "
```

rkaczorek, 2022-09-06:
You can try debugging this problem with `gdb --exec kstars --args run`
Make sure that kstars-bleeding-dbg in installed on your system

Akinetopsia1, 2022-09-06:
thanks @rkaczorek 
here is the output of a test run with only gphoto ccd enabled in the profile, test sequence of 30 images, crashed on the fifth

`astroberry@astroberry:~ $ gdb --exec kstars
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) run
Starting program: /usr/bin/kstars 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xaf673040 (LWP 8844)]
libEGL warning: DRI2: failed to authenticate
[New Thread 0xa9e8f040 (LWP 8845)]
[New Thread 0xa968e040 (LWP 8846)]
[New Thread 0xa8aff040 (LWP 8847)]
[New Thread 0xa82fe040 (LWP 8848)]
[New Thread 0xa7afd040 (LWP 8849)]
[New Thread 0xa6eff040 (LWP 8850)]
[New Thread 0xa66fe040 (LWP 8851)]
[New Thread 0xa5efd040 (LWP 8852)]
qt5ct: using qt5ct plugin
Call to writableLocation without an application-based location.
[New Thread 0xa4ebf040 (LWP 8853)]
[New Thread 0xa403d040 (LWP 8854)]
[New Thread 0xa203c040 (LWP 8855)]
[New Thread 0xa003b040 (LWP 8856)]
[New Thread 0x9d34c040 (LWP 8857)]
[New Thread 0x9bee4040 (LWP 8858)]
[New Thread 0x9b6e3040 (LWP 8859)]
[New Thread 0x9ac88040 (LWP 8860)]
[New Thread 0x9a487040 (LWP 8861)]
[New Thread 0x99c86040 (LWP 8862)]
[New Thread 0x9914c040 (LWP 8863)]
[New Thread 0x96dc4040 (LWP 8864)]
[Thread 0x96dc4040 (LWP 8864) exited]
[Thread 0x9bee4040 (LWP 8858) exited]
[Detaching after fork from child process 8869]
[Detaching after fork from child process 8870]
[Detaching after fork from child process 8871]
INDI::BaseClient::connectServer: creating new connection...
[New Thread 0x9bee4040 (LWP 8873)]
INDI::BaseClient::connectServer: Already connected.
Dispatch command error(-1): Device GPhoto CCD not found
<setTextVector device="GPhoto CCD" name="DRIVER_INFO" state="Idle" timeout="60" timestamp="2022-09-06T23:04:14">
    <oneText name="DRIVER_NAME">
GPhoto CCD
    </oneText>
    <oneText name="DRIVER_EXEC">
indi_gphoto_ccd
    </oneText>
    <oneText name="DRIVER_VERSION">
3.2
    </oneText>
    <oneText name="DRIVER_INTERFACE">
2
    </oneText>
</setTextVector>
[... 102 lines cut ...]
[Thread 0x685fe040 (LWP 8938) exited]
[Thread 0x69860040 (LWP 8937) exited]
[Thread 0x8d249040 (LWP 8943) exited]
[Detaching after fork from child process 8948]
[New Thread 0x8d249040 (LWP 8949)]
[New Thread 0x69860040 (LWP 8950)]
[New Thread 0x685fe040 (LWP 8951)]
[New Thread 0x68dff040 (LWP 8952)]
[New Thread 0x8ea4c040 (LWP 8953)]
[New Thread 0x8e24b040 (LWP 8954)]
[New Thread 0x8da4a040 (LWP 8955)]
[New Thread 0x859ed040 (LWP 8956)]
[Thread 0x8d249040 (LWP 8949) exited]
[Thread 0x8e24b040 (LWP 8954) exited]
[Thread 0x8da4a040 (LWP 8955) exited]
[Thread 0x8ea4c040 (LWP 8953) exited]
[Thread 0x68dff040 (LWP 8952) exited]
[Thread 0x685fe040 (LWP 8951) exited]
[Thread 0x69860040 (LWP 8950) exited]

Thread 24 "kstars" received signal SIGBUS, Bus error.
[Switching to Thread 0x9a487040 (LWP 8882)]
0xb6fb9c1c in memcpy () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
(gdb) 
`

rkaczorek, 2022-09-07:
Could you install indi-gphoto-dbg and rerun the debuging?

Akinetopsia1, 2022-09-07:
i quickly tried and sudo apt installed earlier to the same result and the same lack of information. I'm more familiar with windows debuggers so i'm not sure about the process on linux, but wouldn't i need to compile that debug version from source -- specifically the same 3rd-party source i compiled from initially ?

I also assume i'd have to start this separate indi service process in its own debugger -- From past attempts the first time i start the profile from ekos it asks to restart indi (i have it in autostart), otherwise it never really sucessfully connected.  havent had much time to investigate today, will try tonight.

Akinetopsia1, 2022-09-20:
Just thought i'd make an update: basing myself on @rkaczorek 's work i ended up creating basic flask webapp plugin that doesn't rely on the newer libs. It only displays temporary jpgs for preview purposes but saves .CR3 files internally on the camera, which can then be downloaded on the pi home/astroberry/session/targetname folder once the sequence is done to make it easier to organize.

I don't know how much of this was for nothing, or when a new release can be expected but for the mean time i'll at least be able to do some automated sessions. It's still a bit rough but can share the code after a bit of clean-up if it's ok and that there's anyone's unlucky enough to be in a similar situation to me.

[image]

App is designed to be responsive and is mobile compatible, works on phone and tablets -- which was my main intended use case.

rkaczorek, 2022-09-21:
@Akinetopsia1 it looks promising. I would be more than happy to look at the code when you're ready to share it. I might include it in Astroberry 3.0 as I'm thinking of purely web interface for INDI (as an alternative to EKOS)

damianjableka, 2022-10-12:
Two days ago, after a few weeks of not using, my SD card died. Unexpectedly not as usually in RPi with fsck error but straight on boot there was no reading. Of course I have a lot of backup copy of all my RPis but non with astroberry. So I cooked everything from scratch.
So yesterday was long day
I recompile `LibRaw-0.20.2`,  `indi-core`,  `indi-gphoto`, `indi-eqmod`, `stellarsolver-build` smuthly 
during `indi-sv305` it wass nesseserry to coment out part of code with `status = SVBGetCameraPropertyEx(cameraID, &cameraPropertyEx);` because it generates error, and in fact there was no such function anywhere in code. 
Thanks to @Akinetopsia1 to mention about `extra-cmake-modules` during `sudo apt install libev-dev`
And at the end `kstars`... I tried `--branch stable-3.6.1`, all way down to `stable-3.5.6` every time it was an error around 25% I think that somehow connected with `weather.cpp`, sometimes error, sometimes just hang of all RPi. I almost gave up.
Last try was `kstars` `3.4.3` and it works now, I'm able to run `kstars`, `ekos`, took photos and retrieve it I hope that rest of devices will work in the field.

jfrabell, 2023-04-30:
I know this is a very old thread, but I wanted to throw this on the discussion:

My run through failed to build on the cmake command for the core indi library.  I had to run 

sudo apt-get install libev-dev

Then try agin.

vavallee-wp, 2023-11-24:
So we've had this issue for over 3 years? Are there plans to merge a fix. This is messy.

jackfrost373, 2023-12-30:
Surprised as well to see this issue has not yet been resolved for so long. But Debian bullseye took their sweet time as well, it was just released on October 7th, 2023. I'm curious to see a follow-up release of astroberry soon.

closed 2026-02-11 by rkaczorek