rkaczorek/astroberry-server #60: Unable to connect to Wifi hotspot with Android phone

AmiSMB, 2020-03-16:
I have Astroberry 2.0.1 on a Raspberry Pi 3b but am finding that I can connect to the wifi hotspot with Windows 10, Mac OS X 10.14.6 and Ubuntu 19.10 but not on the device that I wish to use when out which is a Samsung Galaxy S10 on Android 10. I can see the astroberry wifi hotspot and correctly entered in the password as I have on the other devices and it tries to connect for quite a while but ends up with an error unable to connect.

All I see in /var/log/syslog when Android connects and then fails:-

astroberry wpa_supplicant[341]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0

rkaczorek, 2020-03-18:
AFAIK it is related to wpa_supplicant version provided by official raspbian buster version (v2.7).
The latest version of wpa_supplicant is v2.9, which apparently solves the issue. See https://raspberrypi.stackexchange.com/questions/107188/android-wont-connect-to-raspi-access-point

AmiSMB, 2020-03-19:
I have made the following script to build and install the latest wpa_supplicant on a Raspberry Pi running Astroberry. 

Scp the attached file wpa.txt to Astroberry and the ssh in.
Do the following commands:
 mv wpa.txt wpa.sh
 chmod +x wpa.sh
 ./wpa.sh

It now works great now for me and I hope this helps someone else in the same situation.

[wpa.txt](https://github.com/rkaczorek/astroberry-server/files/4357217/wpa.txt)

drakth, 2021-01-19:
> I have made the following script to build and install the latest wpa_supplicant on a Raspberry Pi running Astroberry.
>
> Scp the attached file wpa.txt to Astroberry and the ssh in.
> Do the following [...]

Thank you so much, this worked perfectly!

SankaCoffy, 2021-08-21:
What do you mean by "Scp the attached file wpa.txt"
Don't understand what the check mark is on the last line just before the wpa.sh and how do you create it on the key board?.  Never have seen that before
Think I understand remaining.  (Still new to linux based systems, sorry)
Thank you.

Agwising, 2022-02-23:
Where do i find the wpa.sh file?

rkaczorek, 2022-02-23:
It's attached to the post above

Agwising, 2022-02-24:
> 

I can only see wpa.txt

Agwising, 2022-02-24:
I think i got it. Just SCP the wpa.txt then run the commands as described. However i would still need to connect the Astroberry to the internet via ethernet since sudo apt-get update will run

Agwising, 2022-02-24:
Still not detecting my hotspot. I noticed that the wpa supplicant files are installed to the same folder where i scp the wpa.txt. which is Documents in this case. Is this ok or should i transfer these files?