rkaczorek/astroberry-server #61: Trying to add ppa repository for focuser driver

mikez104, 2020-03-22:
Trying to run the following command to add a new repository for my Astromechanics focuser.
`sudo add-apt-repository ppa:mutlaqja/ppa`
This command fails with the following error.
```
Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 93, in get_sources
    (self.id, self.codename))
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Raspbian/buster
```
I tried to manually add it in /etc/apt/sources.list.d but this did not work.
Here's the launchpad ppa.
`https://launchpad.net/~mutlaqja/+archive/ubuntu/ppa`
Could anyone help?

rkaczorek, 2020-03-26:
First, the repository that you want use is for Ubuntu, not for Raspbian. This is why you get "Error: could not find a distribution template for Raspbian/buster". It is not supported on Astroberry, which is based on Raspbian distro. Using it may result in unpredictable results.
Second, indi-astromechfoc driver is available in Astroberry. If you cannot find it you need to install it by running:
`sudo apt install indi-astromechfoc`

mikez104, 2020-03-27:
Thanks so much for your reply.  I mistakenly thought that Raspbian was based on Ubuntu.  Thanks for pointing me in the right direction!

closed 2026-02-11 by rkaczorek