rkaczorek/astroberry-server #62: Cannot connect telescope in Kstars

jakugab, 2020-03-25:
Dear Radek!
First of all, I would like to say a great thanks for Astroberry Server and for sharing with all of us.
I downloaded Astroberry Server 2.0.1. image to my RPi 3B+. When I tried to connect to the telescope in Kstars, I got an error message: cannot connect to telescope, /dev/ttyUB0 is used by another driver or process. As I figured it out, the /dev/ttyUSB0 port is used by gpsd.  I do not use GPS dongle yet. In an earlier version of Astroberry (based on Ubuntu Mate)  the scope connected without any problem to this port. The scope controller is an OnStep board. How can I solve this problem? (Sorry, I'm just an end-user without any programming skills :-(  ).
Thanks in advance,
Gabor

rkaczorek, 2020-03-26:
Are you sure that the port is used by gpsd? It could only happen because gpsd takes your mount as a gps device. To confirm that:
1. Disconnect your mount from usb
2. in your terminal run: tail if /var/log/syslog
3. Connect your mount to usb observing the log entries
You should see entries saying usb device is conencted, what port it is assigned etc. If gpsd grabs it there will be additional entries indicating this.
If this is the case you can disable gpsd by running:
`sudo systemctl stop gpsd.service
sudo systemctl stop gpsd.socket
sudo systemctl disable gpsd.service
sudo systemctl disable gpsd.socket
`
This will be a problem if you want to use gpsd dongle in the future.

jakugab, 2020-03-26:
Hi,

Thanks for your response.
I disconnected the mount from usb, ran the command in terminal, and got
this:

astroberry@astroberry:~ $ tail if /var/log/syslog
tail: cannot open 'if' for reading: No such file or directory
==> /var/log/syslog <==
Mar 26 18:16:59 astroberry systemd[1]: gpsd.service: Succeeded.
Mar 26 18:16:59 astroberry systemd[1]: Stopped GPS (Global Positioning
System) Daemon.
Mar 26 18:16:59 astroberry systemd[1]: gpsd.socket: Succeeded.
Mar 26 18:16:59 astroberry systemd[1]: Closed GPS (Global Positioning
System) Daemon Sockets.
Mar 26 18:16:59 astroberry systemd[1]: gpsdctl@ttyUSB0.service: Succeeded.
Mar 26 18:16:59 astroberry systemd[1]: Stopped Manage ttyUSB0 for GPS
daemon.
Mar 26 18:17:01 astroberry CRON[1311]: (root) CMD (   cd / && run-parts
--report /etc/cron.hourly)
Mar 26 18:18:18 astroberry kernel: [ 3104.137911] usb 1-1.3: USB
disconnect, device number 7
Mar 26 18:18:18 astroberry kernel: [ 3104.138627] cp210x ttyUSB0: cp210x
converter now disconnected from ttyUSB0
Mar 26 18:18:18 astroberry kernel: [ 3104.138727] cp210x 1-1.3:1.0: device
disconnected

When I connected the mount to usb, and ran the above command I got this:

astroberry@astroberry:~ $ tail if /var/log/syslog
tail: cannot open 'if' for reading: No such file or directory
==> /var/log/syslog <==
Mar 26 18:26:04 astroberry mtp-probe: bus: 1, device: 9 was not an MTP
device
Mar 26 18:26:04 astroberry systemd[1]: Starting Manage ttyUSB0 for GPS
daemon...
Mar 26 18:26:04 astroberry gpsdctl: gpsd_control(action=add,
arg=/dev/ttyUSB0)
Mar 26 18:26:04 astroberry gpsdctl: reached a running gpsd
Mar 26 18:26:04 astroberry systemd[1]: Starting GPS (Global Positioning
System) Daemon...
Mar 26 18:26:04 astroberry systemd[1]: Started GPS (Global Positioning
System) Daemon.
Mar 26 18:26:06 astroberry ModemManager[321]: <info>  Couldn't check
support for device '/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2':
not supported by any plugin
Mar 26 18:26:07 astroberry kernel: [ 3572.421397] pps pps0: new PPS source
usbserial0
Mar 26 18:26:07 astroberry kernel: [ 3572.421769] pps pps0: source
"/dev/ttyUSB0" added
Mar 26 18:26:07 astroberry systemd[1]: Started Manage ttyUSB0 for GPS
daemon.

I disabled the gpsd daemon as suggested, the result was this:

astroberry@astroberry:~ $ sudo systemctl stop gpsd.service sudo systemctl
stop gpsd.socket sudo sytemctl disable gpsd.service sudo systemctl disable
gpsd.socket
[sudo] password for astroberry:
Failed to stop systemctl.service: Unit systemctl.service not loaded.
Failed to stop stop.service: Unit stop.service not loaded.
Failed to stop sytemctl.service: Unit sytemctl.service not loaded.
Failed to stop disable.service: Unit disable.service not loaded.
Failed to stop systemctl.service: Unit systemctl.service not loaded.
Failed to stop disable.service: Unit disable.service not loaded.

At this point, if I start Kstars, I can connect to the mount successfully,
everything works fine. But, after rebooting the system, running Kstars,
starting my gear in Ekos (mount controller is powered on), I cannot connect
to the mount (error message in Kstars is: port /dev/ttyUSB0 is used by
another driver/process). After disabling the gpsd service with the above
mentioned command again, the connection can be established with the mount.
If I understand it right, the /dev/ttyUSB0 port is switching between the
gpsd and the CP2102 USB-UART converter (this connects the telescope
controller to the RPi). Why must I disable gpsd service after a reboot
again?

Thanks in advance, Gabor

Radek Kaczorek <notifications@github.com> ezt írta (időpont: 2020. márc.
26., Cs 17:57):

jakugab, 2020-04-07:
Hello,
Problem solved by editing the /etc/default/gpsd file, changing "true" to "false". 
Thanks for your help.

closed 2020-04-07 by jakugab