rkaczorek/astroberry-server #63: Should firecapture be added to default installation?

GrabzIt, 2020-03-26:
It seems firecapture is now compatible with the pi. So would it be a good idea to have this on astroberry by default it seems that many people are manually installing it.

It would probably be a good tool to have on there as firecapture is great for planetary.

rkaczorek, 2020-03-26:
This is a great tool indeed. However it cannot be added to astroberry by default as for now. Distribution is prohibited by license agreement. I will try to get the permission of the author.

rkaczorek, 2020-04-04:
FireCapture in the house! Now you can install it from official astroberry repo.
Run: sudo apt install firecapture

closed 2020-04-04 by rkaczorek

GrabzIt, 2020-04-04:
That's awesome thanks!!

spoupart, 2020-05-27:
I have this error : E: Unable to locate package firecapture 
when i run: sudo apt install firecapture

ManuelAstro, 2020-06-21:
Thanks so much for your work @rkaczorek, I’ve installed the package from the repo but I can’t find the app to be launched.