rkaczorek/astroberry-server #65: How to align mount?

yesyesuk, 2020-04-10:
Hi. 
I'm new to Astroberry and KStars/Ekos.
I've used my telescope with EQASCOM before but haven't used it for several years. Now I want to start using it again with Astroberry on a RPi 3 B+. The telescope is in my own observatory.

I managed to connect all my devices (HEQ5 mount, Canon EOS 1000D, QHY 5-IIL for guiding, DIY motor focuser simulating a Moonlite focuser).

My mount is in an unknown position. How do I make sure the software knows where the telescope is pointing? Are there any tutorials how to do that? Is it enough to just slew anywhere in the sky (manually via software) and then take a photo, solve it and sync the mount?

Thanks!
Chris

rkaczorek, 2020-04-10:
Your last sentence is the answer to your question ;-)
Take a photo, solve it and sync the mount.

yesyesuk, 2020-04-11:
Should it be so easy? :D 
Thanks a lot! I will try that the next clear night.

And thanks for making Astroberry available for everyone! Much appreciated.
Chris

rkaczorek, 2020-04-13:
Just to makes sure - this will align your mount for goto only and does nothing about polar alignment. If you're interested in the latter go to Ekos/Align tab and run Polar Alignment procedure.

yesyesuk, 2020-04-14:
Thanks. Yes, I wanted to align the mount so it knows where it is pointing (for goto), not polar alignment.
I tried that last weekend and it worked perfectly.

I've noticed the Polar Alignment tab in Ekos last night. I will try that when I get another clear night to see if my polar alignment (from 4 years ago) is still fine.

I'm really loving this software. It makes an imaging session sooo much easier.

This is probably not the right place for any Ekos issues / suggestions?

rkaczorek, 2020-04-14:
I'm glad it works for you.
You can submit core INDI related issues here: https://github.com/indilib/indi/issues
INDI 3rdparty drivers here: https://github.com/indilib/indi-3rdparty/issues
The best place to report and discuss KStars/Ekos issues is https://indilib.org/forum/
All the best!

yesyesuk, 2020-04-15:
That is service!! Thank you very much!! ;-)

closed 2020-04-15 by yesyesuk