rkaczorek/astroberry-server #66: GPSD and indi driver

jamesbaillies, 2020-04-10:
First of all a big Thank You for building Astroberry! It really is a wonderful resource for amateur astronomers. I've been trying to figure this problem out for a couple of days now and I have yet to find a solution. The crux of the problem is that, when I have a USB gps device plugged into my PI4, the indi gpsd driver displays an incorrect longitude. The longitude should be -114:04:59 and it displays 245:39:54. I've run KSTARS from the command line and the output there displays the correct GPS location. The GPS device is a u-blox 7.

rkaczorek, 2020-04-10:
All is fine, even if it seems incorrect. INDI Longitude is 0 to 360 East+

jamesbaillies, 2020-04-11:
Oh I see. It's confusing none-the-less. Thank you for your quick response!
All the best,

James

rkaczorek, 2020-04-13:
It is legacy of KStars which also uses this approach for longitude.