rkaczorek/astroberry-server #67: Camera on RPi +3B

M31Fan, 2020-04-14:
Thank you for creating Astroberry! 

I'm trying to get the autoguiding to work with a regular USB webcam or Raspberry camera module.
I'm getting massive lags (5 to 10 sec) and also extremely bad frame rates (about 1fps).
Raspstill works fine. Raspvid does horrible... Are there any sort of lib, or codecs missing? 

INDI has no problems in recognizing either camera (V4l2 CCD), so once the camera is working well with the Astroberry, I'm confident PHD2 will also work well with it.
Many Thanks!

rkaczorek, 2020-04-14:
Does it mean that you get lags using V4L2CCD driver?
What is the result of using an independen webcam app such as cheese? (sudo apt install cheese)

M31Fan, 2020-04-14:
Hello Rkaczorek,

It's a bit more than just a lag.

I reflashed the SD card, run update and upgrade to ensure I had a fresh
start.
I then installed cheese and tried various combinations. Results:
- Cheese with any of the cameras: nothing. Simply doesn't detect them.
"device not found"
- I was able to obtain some images though PHD2, through INDI, but the
images are very bad. PHD2 crashes a lot.
- Raspistill works well (with both cameras)
- Raspivid only works in creating H264 files, but when I replay them with
VLC the quality is appalling, frame rate is around 1 fps and the duration
is aproximately half.

 Any clues, or additional tests, to track and resolve this issue?

Many thanks,
Tiago

M31Fan, 2020-04-14:
I would like to add that I installed Webcamoid and it was able to detect and identify my USB cam (located at /dev/video0). If I keep it with the initiall resolution of 1920x1080 I get about 10 sec of lag. Reducing to the lowest resolution in webcamoid (1280 x 720) lowers the lag and improves the fps, but still unusable. 

Many Thanks!
Tiago

rkaczorek, 2020-04-14:
Could you reproduce this on standard Raspbian Buster for Raspberry Pi Foundation?

M31Fan, 2020-04-14:
I’ll give a try to reproduce it. Thanks!

M31Fan, 2020-04-16:
Hello again, 

Sorry it took longer to reply... Kids are getting restless 😬 

I followed a suggestion of a friend, got one of the test files I had crated and tested in a separate computer: It looks perfect. 
So the issue is not capture but the playing or the streaming. 

Meanwhile I found this thread, where the equipment, method and problem are exactly what I was facing. 

https://www.raspberrypi.org/forums/viewtopic.php?t=225073

Seems the bottleneck is the VCN. At least 1.5 years ago it was. 

If I understand correctly, VCN allows us to a browser to access the headless Raspberry, and I love that feature! 

On one side this is good, as it means PHD2 won’t see the problem I was seeing. 
On the other hand I’ll need to find a different approach to stream with better quality, as I also  wanted to use the camera to do my polar alignment without breaking my back. I’m considering netcat. 

What are your thoughts on this? 
Any way to improve VCN performance, or simple use it as it is for AP, and use netcat to PA? 

Many Thanks!! 
Tiago

rkaczorek, 2020-04-17:
You can access Astroberry Server via VNC or via web browser. Both ways will stream full desktop to your PC in hight resolution. I'm doing planetary imaging up to 120 fps without any problem and I use browser connection to Astroberry Server.
For PA you can use Align in Ekos, which provides very good PA module. No need for a cam in polar alignment scope as the module uses main scope for PA. Give it a try!

M31Fan, 2020-04-21:
Hello Again,

Thanks for the tip on the PA. I have not yet invested in a dedicated camera
for the main scope, and I'm simply using a DSLR (with a remote shutter
controler). I'll check if I can use the guidescope for it.

Meanwhile, regarding the camera viewing performance, I made quite a few
tests and here are my findings:
- With Raspian Buster, connected directly to an HDMI TV: Cheese doesn't
recognize the camera; VLC also doesn't as it stalls completely. This is
exactly the same behavior wth Astroberry.
- Mplayer and FFplay did provide some sort of a workable solution, via
browser, albeit not perfect, it's still usable. It was the same in Raspbian
buster and Astroberry, so the last tests were only in Astroberry.
- Depending how I start the system, /dev/video0 can be Raspicam, or the
webcam connected via USB. I opted to only connect the USB after the
start-up to have Raspicam on /dev/video0, and therefor the USBcam is
assigned to /dev/video1.
- "Mplayer tv://"  shows a pop-up window with the camera view. This allows
me to check the camera that is connected to /dev/video0.
- ffplay can also give me a pop-up window of either the cameras
(/dev/video0 or /dev/video1). Using -an (no audio) get's slightly better
image.
- I'm now able to see both cameras simultaneously, without massive issues,
but a few things have quite an impact on lag and usable resolution:
--  Resolution of the desktop, as VNC has to send it over the network (I
reduced it to 1024p)
--  Resolution displayed by the cameras: I kept them at 640x480@15fps, as
this is usable.
--  Type of network connection (2.4Ghz vs 5Ghz). Using the hotspot set
manually at 5Ghz I get better network performance and with it reduced lag
and better fps.
--  With both cameras view on, the CPU gets pretty busy, but I have no idea
if the GPU is doing any work. Do you recommend that I invest on a RPi 4?

Potential ways to improve the camera's view:
- reducing color to grayscale (I tried several commands, but with mixed
results, most rather complex and without real benefit. I reverted to
simpler standard colors).
- using gstreamer or netcat, but all the tips I could find were several
years old and no longer work. Unfortunately my knowledge is very limited to
work around these issues.
At the end of this rounds of tests and attempts I was not able to find any
way to have such a high performance as you described. I welcome any tips ;)

Essentially, for now it seems that the lag and low fps come from the fact
that I'm using it remotely (local network), so my hope is that PHD2 won't
be afected by it. I'll need to find the time to try it.

I thank you your time and effort in creating this awesome tool!
Many Thanks!
Tiago

rkaczorek, 2020-08-09:
Raspberry Pi Cam driver has been added. Please upgrade your system and report back.

yesyesuk, 2020-08-16:
> 
> 
> Raspberry Pi Cam driver has been added. Please upgrade your system and report back.

Does this also support the new High Quality RPi camera?

rkaczorek, 2020-08-23:
Yes