rkaczorek/astroberry-server #68: Newbie recurring problem - initial screen resolution

cstachon, 2020-04-23:
Just a thought from helping people but the most common problem for newbies is the screen res. A lot of people do not have a device that can display the whole screen via VNC (mainly Web NOVNC).
Any chance your image could start with a smaller Res and just some pointers to increase the res later or for people with better screen resolution.  

For newbies it just give a bad initial impression (for some) of being a problem. So it works straight out of box.

Enhancement :-)

rkaczorek, 2020-04-24:
Thanks for the suggestion.
Normally, resolution is adjusted to user's screen resolution detected via HDMI but this results in very small screen if system is running headless. To address this issue resolution is hard coded now in /boot/config.txt and set to 1920x1080 (full HD). There are two options:
1) revert to autodetection so users will have to set proper resolution on their own
2) enable autoscaling in web interface (noVNC)

The former can be handled by raspi-config or Raspberry Pi Configuration (GUI), the latter does not help much for accessing system via VNC :-/
What do you think would be the best approach?

cstachon, 2020-04-24:
Its very hard to cater for every level especially as many have never used Linux etc before (they mainly come from Windows) but as with all things first impressions tend to stick.
The ideal way would be auto Res but thats not really feasible I guess.
How about a first time pop up that enables the user to change the Res (some will still screw it up I guess)  or you have 2 or 3 backup configs (so std copy of config.txt but with different Res in each config.low,config.med,config.high) which the user can choose and then reboot.
Once they become more aquainted they will know how to change the Res anyway.
I realise its difficult and it does depend on who(level of user)  you want to use your great set up.
I had similar problems with https://astrovideoforum.proboards.com which you have already met :-)

Keep Safe Radek in these weird times!

rkaczorek, 2020-04-24:
It's not easy for a mountain goat to accompany a climbing man, as Stanislaw Lem wrote in one of his books. The goat gets much more tired than a man, because it has to make many jumps up and down to keep up with a slow man ;-) I will think it over what is the best way to approach it. Thanks for the suggestion anyway.
All the best to you too Stash!

cstachon, 2020-04-24:
LOL - very true. All I can do is ask :-)

rkaczorek, 2020-04-25:
I didn't mean you ;-) It's rather me having difficulties to imagine how inexperienced users do things. That's why UX experts are making their money ;-)
Let's keep the issue open so I remember about this enhancement when preparing next Astroberry release.

cstachon, 2020-04-25:
Most are Windows users who cant think - just complain LOL    Do you hear if there is any chance that Ekos will split and be stand alone. There are some features I like in Ekos - for my DSLR Live view but to be frank I use CCDCIEL (On Windows) because the Sequencer is simple,works 99% of the time, like me but it doesn't have LIVE VIEW :-(

rkaczorek, 2020-04-25:
I don't think it will ever happen. Ekos is way too much integrated to KStars. I believe it would be good idea to split the two but it's a major challenge and huge effort. Technically it is possible, especially thanks to D-BUS API that is provided by both KStars and Ekos. So far there's no a single brave man to accept this challenge though.

GrabzIt, 2020-04-26:
I just came across this and feel like a right idiot. I have been scrolling around for the last week I didn't think there was this option in novnc.

Maybe all it needs is a mention as the second thing to do on the getting started guide. Although I think it is only useful on largish tablets and above. I tried it on my phone and it's unusable.

rkaczorek, 2020-04-26:
Added as a note to How to use it and FAQ

stevecoh1, 2022-05-31:
> Added as a note to How to use it and FAQ

Would you be so kind as to provide a link to this information? This issue drove me crazy last night. I finally had all my sh*t together (I thought), dragged my scope out for observing, using the NoVNC connection and couldn't control the windows. They were too big for the screen, no way to scroll it and found it difficult to even close windows, much less shrink them. Would a VNC client give better results?  Otherwise, I very much like the convenience of NoVNC. I'm a Linux user, by the way and this defeated me.

rkaczorek, 2022-05-31:
You can use `sudo raspi-config` in terminal or autoscale in browser, see [here](https://www.astroberry.io/docs/index.php?title=Astroberry_Server#Screen_resolution_does_not_match_my_display._How_can_I_fix_it.3F)

stevecoh1, 2022-05-31:
OR, adopt the much simpler solution I found:
1) Open Astroberry NoVNC server in a separate browser window
2) Go full screen on that window.
Problem solved. 

Maybe this helps the next dummy! :-)

Thanks for Astroberry, by the way!