rkaczorek/astroberry-server #69: Kstars crashing repeatedly

M31Fan, 2020-05-04:
Hello! 

My Kstars is crashing every time I try to use Ekos. INDI is able to connect to Mount, QHY guide camera, Cannon, and MyFocuserPro2. 
Before I could finish the focus process the Kstars/Ekos crashed. Astroberry remained on, and upon restart of Kstars/Ekos INDI was still active. 

I also did sudo apt update, sudo apt upgrade and restarted kstars, but the outcome was the same. 

Please advise on how to track and resolve this. 
I'm now going to try with PHD and other software to see if this is kstars only. 

Many Thanks for your great help!

rkaczorek, 2020-05-05:
Please enable logging for INDI and your devices (see Ekos main tab Options at bottom right). You can upload the log file(s) here so I can look at it. Which version of Raspberry Pi do you use? What is the model of Canon and QHY? Do you connect the devices directly to RPi or via USB hub?

M31Fan, 2020-05-15:
Hello!

Apologies for the late reply.
I have tried to find a way to reproduce the failure but wasn't able to.
I did remove some software packages that were recommended (after sudo apt
update) and it seemed to help, however, I also found that an occasional bad
contact between cable and mount control gives me contact errors
which occasionally causes the Kstars to crash. So perhaps that was
the cause.

Whenever this bad contact happens, after re-checking the cables (or
restarting my camera after swapping battery), I always have to reboot Rpi
to be able to connect the equipment with INDI (even restarting INDI is not
enough). A USB reset (or refresh) would be a great help.

By the way, every time I connect my EQmount, I always have to go to the
terminal and type: sudo systemctl disable gpsd.socket
I've meanwhile created a bash file and just run it right after turning the
system on. Then I have no problems.

I just finished a 2h session on M81. Just one restart at the very beginning
because of my mount's cable connection, but after that no glitches.

Since the probable cause of this crashing seems to be down to my cables, I
guess this topic can be closed.

Many Thanks for your continued effort in making this such a great system!
Tiago

rkaczorek, 2020-05-15:
Thanks for an update Tiago. Closing for now.

closed 2020-05-15 by rkaczorek