rkaczorek/astroberry-server #70: Unable to connect a canon EOS 1000D

MrPolnareff, 2020-05-08:
Hi,
I'm struggling to get a canon EOS 1000D working in Ekos. Astroberry server is running on a raspberry pi 4 with 4Gb RAM. The installation is fresh and i've done an apt update/upgrade. I've created a profile in Indi Webmanager with the Canon DSLR driver. Then in Ekos, the CCD fields is configure for Canon DSLR also (see the picture).
[image]

I get the following error.
[image]

The Camera is plugged and powered. I've also tested the Gphoto Driver with not more results (exept that one times it has worked and i've managed to take pictures). I get the following error with Gphoto.
[image]

The camera is in manual and bulb mode. Mirror locked up and automatic shutdown are disable.

Thanks for your hard work and this beautiful project.

MrPolnareff.

yesyesuk, 2020-05-08:
Can you try with a different USB cable? 
I have the same camera and it works absolutely fine for me (but on an RPi 3B+)

MrPolnareff, 2020-05-09:
Yes i have tried with 3 differents usb cable that i've checked i can transfer data beforehand. I've also tried with a RPI 3b but i got the same problem. I'm started to think that the probleme might be on the camera side.

MrPolnareff, 2020-05-09:
I've managed to connect with ekos by resetting the camera's settings. Sometime i can't start a sequence or a preview but restarting ekos seems to fix the issue. I also use the Gphoto driver.

closed 2020-05-09 by MrPolnareff

rkaczorek, 2020-05-09:
Always make sure that camera is not mounted as an USB external drive
Also you can try GPhoto driver instead of Canon

MrPolnareff, 2020-05-09:
i've never noticed the camera being detected as a usb drive (No arrow to unmount it in the top right corner). But i'm not sure that it is not happening. How to notice it ? And maybe prevent raspbian from mounting it ?

rkaczorek, 2020-05-09:
This is disabled in Astroberry so I added this comment "just in case".
The ultimate way to verify this is to run mount in terminal and check if the camera is mounted as a drive.

MrPolnareff, 2020-05-09:
Ok thank you for the information. 
Thank you for your work.