rkaczorek/astroberry-server #71: arm64.aarch64 repo

gutaker, 2020-05-09:
Since we already have 64bit Debian for raspberry pi 4/3, it would be great to have your repo conform to 64bit.

[rpi-img-builder](https://github.com/pyavitz/rpi-img-builder)

laheller, 2020-07-06:
Hi @rkaczorek @gutaker 

As the title says, it would be great to have also 64bit packages in the repo, because now the command:
`sudo apt update
`

returns:
`N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://www.astroberry.io/repo buster InRelease' doesn't support architecture 'arm64'
`
BR,

Ladislav

gutaker, 2020-09-17:
A little BUMP as this topic is hot on raspberry pi forums [here](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=275372&sid=1fff34a3b74bab3e5dd0af7c0666a8f5&start=325) and we can already download RaspiOS64 [here](https://downloads.raspberrypi.org/raspios_arm64/images/).

laheller, 2020-09-20:
> 
> 
> A little BUMP as this topic is hot on raspberry pi forums [here](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=275372&sid=1fff34a3b74bab3e5dd0af7c0666a8f5&start=325) and we can already download RaspiOS64 [here](https://downloads.raspberrypi.org/raspios_arm64/images/).

Additionally now we have 64bit Ubuntu Mate downloads:
https://ubuntu-mate.org/download/arm64/

gutaker, 2022-02-03:
Official Raspberry Pi OS 64Bit is already [here](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit)... so maybe this will speed things up?

cloudwitch, 2022-07-20:
I started working on reverse engineering a 64bit version today since there is not a lot of traction towards 64bit.

If there is an "official" effort, I'd like to pitch in and save myself many hours of work.

rkaczorek, 2022-07-21:
90% of packages are already compiled for arm64. To be released this summer

jayjar291, 2022-09-04:
what is the status of this?
I tried to install but I am getting this error
```
sudo apt install astroberry-server-full
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 astroberry-server-full : Depends: astroberry-server-wui but it is not going to be installed
                          Depends: astroberry-server-sysmod but it is not going to be installed
                          Depends: indi-full but it is not installable
                          Depends: kstars-bleeding but it is not installable
                          Depends: gsc but it is not installable
                          Depends: phd2 but it is not installable
                          Depends: phdlogview but it is not installable
                          Depends: oacapture but it is not installable
                          Depends: firecapture but it is not installable
                          Depends: indi-astroberry-diy but it is not installable
                          Depends: indi-astroberry-piface but it is not installable
                          Depends: indi-weather-mqtt but it is not installable
                          Depends: indiwebmanagerapp but it is not installable
E: Unable to correct problems, you have held broken packages.
```

laheller, 2022-09-05:
> what is the status of this? I tried to install but I am getting this error
>
> ```
> sudo apt install astroberry-server-full
> Reading package lists... Done
> Building dependency tree... Done
> Reading state [...]

Means not everything is compiled yet for arm64. You have to wait...

rkaczorek, 2022-09-05:
astroberry-server-full is not ready yet, but... You can add the repo to fresh install of Raspberry OS 64bit (Bullseye) by running:
```
wget -O - https://www.astroberry.io/repo/key | sudo apt-key add -
sudo su -c "echo 'deb https://www.astroberry.io/repo/ bullseye main' > /etc/apt/sources.list.d/astroberry.list"
```
Then run `sudo apt install indi-full kstars-bleeding phd2` to get core apps installed

This is not official yet, so test and report back how it goes.

laheller, 2022-10-17:
@rkaczorek 

So far it works perfectly! I have 64bit packages installed on my RPI:
[image]