rkaczorek/astroberry-server #73: headless-unable to connect to Pi4-any method

jamesbaillies, 2020-05-15:
Hi Radek. First of all, thank you for making astroberry available to the astronomy community. It really is fantastic. After many cloudy nights this spring I was finally able to get out and give the package a test drive and it ran flawlessly all night long. 
Here is the one issue I'm having that I have tried to solve but with no success. Occassionally I can't connect to my headless Pi by any means. Wifi, Ethernet, Ethernet direct connection from a computer that I have configured to work, all of them do not work. The only way to reconnect is to unplug the Pi and plug it back in again. Obviously I don't like to do this as it may corrupt my filesystem. Have you come across this problem before? 
I'm running the latest kernal: Linux astroberry 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux

rkaczorek, 2020-05-15:
Hi James, this must be some kind of hardware issue.
First, I would check power supply. Second, I would make sure that CPU is not overheating (do you use passive or active radiator?). You can check temperature by running in your terminal:
`/opt/vc/bin/vcgencmd measure_temp`
It would be also good idea to reproduce this issue while connected to a display over HDMI (BTW this way you can also check power supply - yellow lightning in top right corner will indicate undervoltage). I don;t think

jamesbaillies, 2020-05-15:
Thank you for your reply. I'm powering my rig from a 12v 30a supply which feeds a fused distribution box. All of my gear is plugged into this box. The Pi itself takes a fused feed (7.5a) from the box and is plugged into a 5v-120v 18w cigarette lighter style car power accessory. I have never seen a yellow lightening bolt appear on the Pi GUI. The temp is usually in the low 40's C so I don't think that's an issue. I'm passively cooling the Pi with a Flirc Case. I did see a posting [here](https://www.raspberrypi.org/forums/viewtopic.php?t=272918)which alludes to the issue but I'm not sure if it pertains to my situation. I will try using the AC power supply to see if the problem persists.

rkaczorek, 2020-05-15:
OK, so it looks like neither power or temperature cause this instability then. The only h/w that can be failing then is microSD card, which somehow hangs operating system after some time. Option 2 is you use 2,4GHz wifi connection, which can get in conflict with USB 3.0 bus (known issue with RPi 4). It does not impact wire connection though. In the former case check the card and/or observe the system connected to a display over HDMI. In the latter case switch to 5GHz mode using wifi.
I would not follow the case described in the linked posting. A user is messing with operating system, not really understanding what he is doing (see disappearing eth0 interface, which does not disappear by the way, but is renamed to enx0100003c6bb5, which is clearly stated in logs he's posting himself). If there's no other way you can try using a workaround described in this thread [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=272494&p=1652849#p1652849) but this is not a solution.

jamesbaillies, 2020-05-15:
Interesting...I wasn't aware to the 2.4ghz wifi problem. I was using it because, and correct me if I'm wrong, I thought there was a little better range on 2.4 ghz. But I'll switch to 5 ghx and see if this helps. Many thanks! Oh and I don't think it is the card because I have two and the same problem manifests. Have a great weekend!

rkaczorek, 2020-05-15:
2.4GHz has better range indeed, however it's slower. Unlike 5Ghz... Try it out and let me know how it goes.
BTW. 2.4GHz conflict with USB 3.0 is an inherent design flaw in RPi so it will not be fixed by software updates, so the only hope is another h/w version which manages interference properly.

jamesbaillies, 2020-05-31:
> BTW. 2.4GHz conflict with USB 3.0 is an inherent design flaw in RPi 4 so

Hi Radek,
I have been meaning to say thank you for your suggestion to switch to 5Ghz
wifi on my Astroberry to solve the connection problem I was having. That
did the trick. I also wanted to ask if the new 8Gig Pi will be
advantageous for the Astroberry? And will the switch to a 64 bit raspberry
pi os affect the Astroberry?

Best regards,

James

rkaczorek, 2020-06-01:
I wouln't expect too much from 8GB RAM on new raspberry pi. Performance it's gonna be very similar to 4GB RAM. It might change whe 64 bit OS is ready. However it's to early for me to comment on 64 bit version of Astroberry.

jamesbaillies, 2020-06-01:
Hi Radek,
Is there a way to gift/donate to help compensate you for all the work you
put into development of Astroberry. It's a wonderful tool and I'd like to
contribute.

Jim

rkaczorek, 2020-06-01:
Jim, it's very kind of you. It is free and open source software by my conscious decision to support humanity development by sharing ;-)
There is a donate button on [the project website](https://www.astroberry.io/#download) to cover technical costs for hosting and maintaining software repository.
All the best and clear skies!

jamesbaillies, 2020-06-06:
Hello Radek,
I bought a new 8 gig Pi and I thought I would just boot the new 8 gig pi from the existing SD card that I'm aready using for my 4 gig PI, but it doesn't boot and I get a black screen instead. Have you ever had this issue? I've tried pinging the IP address but I don't get a response. My understanding was the new 8 gig pi was identical to the old one...or am I missing something? 
Many thanks,

Jim

rkaczorek, 2020-06-06:
It is a new issue with 8GB version. Please follow up #82

jamesbaillies, 2020-06-07:
Many thanks Radek. I'll keep following this thread. Let me know if there is
anything I can do to help.

Jim