rkaczorek/astroberry-server #74: Rubbish Web Astroberry Interface

cstachon, 2020-05-16:
Now that we have such a "Professional" Web page for Astroberry main page  - a new request is to do the same for Astroberry Indi interface  LOL

Your fault for creating such a good interface - well done - was it by hand design or an auto job!

rkaczorek, 2020-05-16:
Well, this is on my list for years now. Maybe some day

Lezten, 2020-10-06:
I work with an raspi pi 4 and astroberry. Up to now I am very satisfied. But two problems are left:
1. I cannot connect the raspi with my router speedport of german telecom.
2. I have an spectrograph on my telescope. Between telescope and spectrograph is a motor driven flip mirror which is controlled by an arduino. The arduino is with usb connected to the raspi. After 10 frames taken by the scientific camera I want to take a calibration frame the flip mirror must be flipped for. How can I program the observation planer?