rkaczorek/astroberry-server #75: Possible System Admin addition for Windows diehards

cstachon, 2020-05-16:
Radek,

Have you ever your Webmin ?

Maybe something to add in for Windows diehards who dont like doing command line "stuff".

Ok it could be a bit of a 2 edge sword - too much knowledge etc and maybe a hole in security.

Runs OK on RPI3b+ and doesn't seem to interfere with Astroberry in any way so far.

Just a idea/thought! :-)

rkaczorek, 2020-05-16:
I think the solution would rather to escape from an operating system at all and convert everything to a dedicated application. No need for low level configuration and messing around. But I don't think there's apple way in heterogenous astronomy world. Unless apple starts producing mounts, scopes, focusers and all the rest. Obviously with round corners and closed architecture.

cstachon, 2020-05-16:
I will keep digging as WEBAdmin allows modification to show/allow what you want. Users again and again have problems with 
1. Finding log info from Linux 
2. Want to showdown/reboot system from outside VNC - i.e. Web especially when headless
3. See graphical info on Disk usage,Backups etc

Its a two edge sword - applications for Astro as you suggest but when things go wrong they want to get system info easily as per Windows . IMO/IME. Some they can already get !

Its user level not fiddlers or experts :-) 

If I can tailor make Webadmin to show "useful" bits I will be back. :-)

rkaczorek, 2020-05-16:
I will take a look at it. I haven't used Webmin for ages ;)

cstachon, 2020-05-21:
You can switch off most functions and create one specially for ASTROBERRY user with the bare min function (or as you see fit). 

Just need to add your side panels /Novnc to it and you have useful all singing newbie interface IMO :-)
Does seem to interfere with any Astro programs - sorry Apps.

Perhaps just a bit to add to your WiKi to allow users the choice - its not difficult to install/use etc.

Seriously dont spend any time on it - you have more pressing issues to do for the next release  :-)

rkaczorek, 2020-05-21:
It could be good click and go interface for GUI fans. I was thinking of simple dedicated web interface similar to INDI Webmanager. We'll see. Thanks for bring it to the table ;-)

closed 2026-02-11 by rkaczorek