rkaczorek/astroberry-server #77: Notification infrastructure

picciux, 2020-05-19:
Hi Radek,

first, thank you very much for Astroberry, I really appreciate it.
During astrophoto sessions, something can go wrong: it's not an astroberry thing, it's that systems are never perfect. I thought: would be nice if Ekos could notify me when an error occur; and found KStars has a configurable notification system. Good! But... hey... How can I see/hear notifications in a headless system? I should mount a speaker on my tube? I have to vnc it all the time to check problems? Mhm... a remote bluetooth speaker? 
Best of all: receive a notification on my phone! Ok so... push notifications: all notification services are cloud (and internet) based, but often dark skyes are not "connected", at least for me. Found [Gotify](https://gotify.net/) and works perfectly. I'm still doing some tweaks to integrate it in Astroberry, but basically it's there.
One step further. People can have different needs: users with a remote-internet-connected-obsy would obviously benefit from a cloud based push notification service, for instance. My idea is a notification infrastructure for Astroberry:

- a frontend (basically a bash script) to send notification (called by KStars notifications when an event occurs)
- a backend that does the actual notification stuff. Astroberry will provide Gotify as a local/lan/standalone backend. Other backends (for cloud based services, etc.) could be implemented in the future.

A simple config file can be used for choosing the backend and enabling gotify or not (if not needed).

In a few days I think I could provide a tar archive with my Astroberry-Gotify integration, as soon as I finished testing it, for you to evaluate.

What do you think about it?

Thanks for all your work.

rkaczorek, 2020-05-20:
This is really cool idea! Thanks for sharing this. Let me know when you're ready and I will give it a look and build a debian package so it can be pushed to the repository.

rkaczorek, 2020-05-31:
Hi, any update on this?

picciux, 2020-06-01:
Sorry Radek, been busy. 
Hope to send you something tomorrow.

picciux, 2020-06-03:
Almost ready: how to send you files?
Thanks

picciux, 2020-06-04:
Sent email.

rkaczorek, 2020-06-04:
Got it! Thanks Matteo. I will look at it over weekend