rkaczorek/astroberry-server #78: U-blox 7 USB Dongle problem

RolandKol, 2020-05-20:
Hi Radek,
Thanks a lot for your initiative with Astroberry!

Sadly, - I have GPS u-blox 7 problem...
I checked other posts even saw your comment u-blox 7 works well most of the time.

Unfortunately, - not for me.
As I understand, it should work out of the box without any adjustments to GPSD, and it did, but only Once.
I tried different USB slots and GPSMON most of the time shows U-Blox recognized, but extracts strange date from somewhere, - 2019-04-07. It also shows some data stream at the bottom, but no fix.
However, sometimes it shows NMEA instead of U-blox, and in this case it has no date/time value at all.
I checked the dongle on Win10, - it works, and receives the fix even inside the house.
Not sure how to fix...
I tried the steps show below, - ended up re-flashing Astroberry back from scratch as no result.
1) got USB port of the GPS device by: ls /dev (checked the list without the device and with device plugged In). result: /dev/ttyACM0
2) amended /etc/default/gpsd to: DEVICES="/dev/ttyACM0"
3) sudo systemctl stop gpsd.socket
4) sudo systemctl disable gpsd.socket
5) sudo gpsd /dev/ttyACM0 -F /var/run/gpsd.sock
reboot
Also tried to disable Virtual GPS
7) sudo systemctl stop virtualgps.service && sudo systemctl disable virtualgps.service - with and without any adjustments listed above...

no luck

[image]

RolandKol, 2020-05-20:
Update:
I got the fix.. after several reboots and after Enabled/Started virtualgps.
sudo systemctl enable virtualgps.service && sudo systemctl start virtualgps.service

But gpsmon behaves strange:
it always flickers the Current Date (2020-05-20) and the 2019-04-07
Also it shows, data is retrieved via /tmp/vgps,  and not via /dev/ttyACM0 as I saw others receive.
Some progress based on Luck only :)

RolandKol, 2020-05-20:
@

rkaczorek, 2020-05-20:
Please take a look at [wiki](https://www.astroberry.io/docs/index.php?title=Astroberry_Server#Setting_geographic_location)

Virtual GPS has to be disabled. Now it feeds GPSD service parallel to your GPS device. You can check if it is disabled by running `sudo systemctl status virtualgps` and/or starting gpsmon - it should not receive any data with your GPS disconnected.

USB dongles there is no need to edit /etc/default/gpsd - they work in plug-and-play mode. So leave DEVICES=""

After you fix the above, disconnect your GPS device, run `tail -f /var/log/syslog` and connect the device again, observing the logs. Your device should be detected and picked by GPSD automatically. Then just run gpsmon and observe the data stream. If there is NMEA data it works. For the FIX I would test it outside. There are many reports that it does not work inside. I have no idea how Windows does it ;-)

M31Fan, 2020-05-21:
Hello!

WIKI is down :( 

I have the same GPS (U-Blox 7). It's brand new and I just plugged it. The VNC panel (on the left) shows the signal and it's correct. That's all the good news. 

Now my challenge is getting my EQMod connected.
In the past I always deactivated the GPSD otherwise, I would get port conflict that wouldn't let me connect the EQMod.  (sudo systemctl stop gpsd.socket). 
Now, when the GPS is inserted in the USB, the "systemctl stop gpsd.socket" doesn't work, and when trying to start INDI (through EKOS) I get the same mount connection errors as before (can't find a port). Even with auto scan it doesn't connect unless I get the /dev/ttyUSB0 free. 

I've learned that Stellarmate has a port assistant is there any similar tool that may help with this?

Many Thanks!
Tiago

M31Fan, 2020-05-21:
Wiki is now up :)

rkaczorek, 2020-05-21:
Yeah ;) it was down for a while because I have to migrate it to another database. Now it should perform much better. It's already up. Let me know if you see any issues with it.
There is no port assistant in Astroberry, yet. It is quite bizarre with this port conflict. Why don't you try to diagnose it using this approach:
- disconnect all USB devices
- run `tail -f /var/log/syslog`
- connect device after device and observe the log
- also test connecting just a mount, then disconnect it and connect just a gps dongle
Doing the above should give you an idea what port is assigned to various devices.
Also uninstall modemmanager: `sudo apt remove modemmanager`  which can be a pain in the back ;)

M31Fan, 2020-05-21:
I'll give it a try tomorrow, right now I have clear skies to go after M82! 

BTW, I'm now doing PA with Ekos as you recommended a few weeks ago. 
After upgrading for RPi 4 and downloading the astrometry files, the PA works really well. Thanks for the recommendation!

rkaczorek, 2020-05-21:
I'm going after M51 right now ;)

M31Fan, 2020-05-22:
I hope your skies are clearer than mine. Here we have clouds, so I came back to this issue. 

Removing the modemmanager didn't change anything. 
The gpsd.socket conflict seems to be related to the gps panel showing on the VNC. 

If I start the RPi4 with all devices connected, I can easily identify which one is connected where with ls -l /dev/serial/by-id 
My mount is on ttyUSB0, my MyFP2 is on ttyACM1 and the GPS (U-Blox7) is on ttyACM0, so there shouldn't be any conflict. 

If I start Ekos, and connect the devices with INDI, then everything connects, except the Mount. 
I now go back to the terminal and run sudo systemctl stop gpsd.socket. The terminal takes quite some time to react, but when it does I can connect the mount and the GPS remains connected and active on INDI (showing updated values when I refresh it). 
The same can't be said of the GPS panel on the VNC side-bar. That one stops updating after I stop the gpsd.socket. Even the time stopped. 
Somehow I think astroberry is using ttyUSB0, which ends up creating a conflict with my mount. 
I don't know if my mount is the one at fault because I've never succeeded in connecting it to any other port. 

Any helps is highly appreciated! 
Many Thanks!
Tiago

M31Fan, 2020-05-22:
Update: After Kstars crashes, restarting EKOS+INDI allows me to connect everything expect the GPS.... I'm too tired to continue troubleshooting tonight. I'll try some more solutions tomorrow.

rkaczorek, 2020-05-22:
I think we need some detailed diagnosis of the port conflict issue. I can't see any way that ttyUSB0 device is conflicting with ttyACMx devices. As the matter of fact ttyACMx devices might be a problem here, because they take id based on connection order. But... ttyUSB0 is something really strange. How do you connect your mount to Raspberry Pi? What is the cable / serial converter?

M31Fan, 2020-05-24:
Hello! 
Problem solved! 
It was fixed through editing a gpsd related file. A good friend gave me a hand and now it works great all 5 devices connecting right at first attempt.

The solution was inspired on the solution described in the link below, expect for the upowerd part which is not part of our case. 
https://www.jann.cc/2012/08/03/gpsd_and_upowerd_are_blocking_my_serial_port.html 

I edited the /etc/default/gpsd to: DEVICES="/dev/ttyACM0"
I also changed the usbauto to "false". 
Reboot 
and it works!

Of course, I now risk that if I change my gear I may have to reedit the files, but somehow with the previous configuration both the mount (USB via Serial converter) and the GPSD were being associated to ttyUSB0 generating the conflict, despite "ls -l /dev/serial/by-id" clearly showing that gpsd was supposed to be on ttyACM0. 

Anyway, I hope this may help other that may have a similar issue. 

Many Thanks!
Tiago

gaevoc, 2020-08-26:
Hello guys,
recently I've been experiencing several problems with a ublox-7 device. Specifically, it happens that the GPS reports the fix via its led but both gpsd and indi-gpsd don't recognise the fix and keep on reporting the status as _fixing..._
I also implemented the workaround described here above by @M31Fan but it didn't completely solve the problem. Now I must restart the GPSD daemon after the system boot in order to make it recognise the real GPS status.
What puzzles me is that everything worked like a charm til about one month ago but I really can't remember anything special that could have caused this.

Thanks
Gaetano

M31Fan, 2020-08-27:
Hi Gaetano!
I just reflashed my card, and apart from the things I described above, I also had to uninstall modemmanager: (as it was instructed by Radek)

sudo apt remove modemmanager

I'm now back at having everything connected, including gps sinal on Ekos. 
The only thing missing is the GPS on the VNC panel.

M31Fan, 2020-08-27:
Gaetano, 
Since you are using a USB device you need to remove the virtual GPS (as instructed in the AB wiki):

systemctl stop virtualgps && systemctl disable virtualgps

I hope this helps!

gaevoc, 2020-08-27:
> Gaetano,
> Since you are using a USB device you need to remove the virtual GPS (as instructed in the AB wiki):
> 
> systemctl stop virtualgps && systemctl disable virtualgps
> 
> I hope this helps!

Hi @M31Fan ,
Thank you for your suggestions. Virtualgps is already properly disabled, I did right after installing the USB dongle.

bbillp, 2021-05-14:
The U-Blox sensitivity is very poor from inside a structure so using it for Lab work is not very desireable so find another GPS with better sensitivity and/or an external antenna that can be placed near a large window or above the roof tops.  Sometimes our U-Blox takes up to an hour to blink the green light, sometimes it takes a few minutes so its not very desireable.