rkaczorek/astroberry-server #79: QHY5L-II CCD no more recognised

Manu4810, 2020-05-20:
Hi Radek,
After doing some package updates, I have suddenly lost the ability to use my QHY5L-II color camera in Astroberry. The INDI driver does not recognise it anymore. OaCapture show the camera in the device list, but indicates that the camera has no loaded firmware and that fxload cannot be found. lsusb says the device is there (1618:0920). qhy_ccd_test returns no device found. Tried different powered USB hubs, all to no avail.  Strangely, the LED behind the camera is very dim, sometimes completely off while the device is plugged. This camera works fine on ASCOM with strong LED signal.
Any clue on this issue before I desperately re-install and re-configure the full Astroberry package to try and sort it out ?

rkaczorek, 2020-05-21:
LED suggests it is power supply issue. USB ports on RPi have their limits in powering connected devices. First, make sure you use juicy power supply for your RPi. Then, test the camera with all other devices disconnected. Using powered USB hub could be a solution too. Ultimate test would be to install KStars and INDI on a PC and test the camera this way to exclude any software issue.

M31Fan, 2020-05-21:
Team, 
I have the exact same camera and last night run all updates and had no issues. 
I'm using a powerfull USB hub to ensure no power issues. 
Best of luck!

Manu4810, 2020-05-21:
I tried with a "juicy" USB hub, and with all other USB ports left free, but that did not improve at all. And I know for a fact that I managed to have this QHY camera up and running up with astroberry until a few days ago even without a powered hub. As you suggested, I plugged the camera into a PC as a test with Ubuntu 18/04 and all INDI drivers installed. Dim light shows up similar to the RPI, but then the light turns strong when the camera is recognised and connected in EKOS. Everything is working fine on the Linux PC, but still not OK on RPI.
I have made two recent changes that may have had a bad effect: I have installed INDIGO then uninstalled the package and drivers. I also tried to plug and connect another QHY CCD camera (different model). Could there be a conflict of firmware loading ? Is is normal to have no QHY related files in the /etc/udev/rules.d directory  ? All this is very confusing...

rkaczorek, 2020-05-21:
Nope, no qhy related files should be sitting in /etc/udev/rules.d
If there are any you can safely remove them. They can interfere with INDI driver for sure.

Manu4810, 2020-05-21:
OK, then.
I can see a that a qhy rule is available in /lib/udev/rules.d. Hope this is OK. However, the /lib/firmware/qhy folder is empty. Is it normal ?

rkaczorek, 2020-05-21:
Nope, it should contain over 70 firmware files. I think you nailed it. Your firmware files are probably gone after you removed INDIGO. Reinstall by running `sudo apt reinstall libqhy`

Manu4810, 2020-05-22:
Hi Radek,
I re-installed the indi library as you suggested and everything is now OK, I even have both QHY CCD camera working simultaneously ! The INDI package may have been broken somehow. Anyway, thanks a lot ! Next step is in the field. Issue can be closed.

rkaczorek, 2020-05-22:
Great! Clear skies!

closed 2020-05-22 by rkaczorek