rkaczorek/astroberry-server #82: rip 4b astroberry install issue

MLKutzke, 2020-06-06:
The install hangs during the upgrade step - it seems to be waiting for a password to access something associated with zram 0.3.2.1-1 during its post install.

Is this a known issue or should I provide additional info?

I am going to go start clean one more time from the astroberry 2.0.1 image to see what happens.

MLKutzke, 2020-06-06:
And it looks like gpspanel.service has the same issue later as I can see a password agent process running when I ssh into the pi and use ps -ef

rkaczorek, 2020-06-06:
It is the second report of installing on Raspberry Pi 8GB. Looks like this version of Raspberry Pi needs additional tests and possible image upgrade. Stay tuned

MLKutzke, 2020-06-07:
If you need me to try something - or run some commands just ask...

MLKutzke, 2020-06-07:
Update: used astroberry image 2.0.1 image
First boot: got normal resizing partition/ restart
Started up with blank screen after normal boot screen rainbow, 4 rasp and spa lash screen
Used ssh connection to confirm up and running
Ran raspi-config to try and restart from command line but
Second boot next got back to the spa lash screen and froze there - but could still ssh to it externally
Tried to use apt to update/ upgrade and it froze at 59% during the upgrade of astroberry-server-wui at what looks like the first ‘Removing ‘diversion of /boot/kernel.img to /user/share/rpikernelhack.img by rpikernelhack’
TMI?  Need more info? Just ask...

MLKutzke, 2020-06-07:
Stuck on sync command in the kernel postinst after kernel move

rkaczorek, 2020-06-07:
Apparently this issue impacts any system released before Raspberry Pi 8GB. I'm waiting for my 8GB version now. Astroberry image can be rebuilt but need to find a way to upgrade existing version.
See [here](https://youtu.be/6EQFpvLGE9Q)

rkaczorek, 2020-06-10:
I have identified the source of the problem for Astroberry 2.0.1 not booting on Raspberry Pi 4B **8GB**. The primary problem is caused by zram swap (enabled by default) which throws kernel internal error 'oops 207'. The source of this problem is not known yet. As for now zram is disabled by default. If you know what you are doing, you can re-enable it by running `sudo systemctl enable zramswap.service`

**Existing users**
If you already use Astroberry 2.0.1 on other Raspberry version, **disable zram swap** before swaping microSD card to Raspberry Pi 4B **8GB**. Run the following command to disable zram swap `sudo systemctl disable zramswap.service` and reboot. Optionally, run `sudo apt update && sudo apt upgrade` to get it fixed for you.

**New users**
If you just got Raspberry 4B **8GB** and downloaded Astroberry 2.0.1, you will have to download **Astroberry 2.0.2** version. Optionally, if you have other version of Raspberry, you can use Astroberry 2.0.1 image and fix the problem just like existing users.

closed 2020-06-10 by rkaczorek