rkaczorek/astroberry-server #83: Please replace legacy NexStar Evolution driver with Celestron AUX

jochym, 2020-06-08:
The new Celestron AUX mount driver is a direct continuation of the NexStar Evolution driver done by the original author and (me @jochym) and the author of the serial connection extension (@fabriziop).
The CAUX driver is a full replacement of the NSEvo driver and is already included in the Jasem  nightlies PPA. Please, replace NSE by CAUX in the astroberry repo. Appart from the extended connectivity (direct serial, HC serial/usb, WiFi), there are also some bug fixes there and the NSE is not going to be developed any further (not even with bug fixes).

rkaczorek, 2020-06-08:
Thanks for the report. The driver will be replaced at the next release planned end of week.

jochym, 2020-06-08:
Great!

rkaczorek, 2020-06-16:
Done

closed 2020-06-16 by rkaczorek

luzik, 2020-11-29:
Kstars give me an error. "Nie znaleziono sterownika indi_celestron_aux w systemie. Prosz? upewni? si?, ?e pakiet, który dostarcza plik wykonywalny 'indi_celestron_aux' zosta? zainstalowany." -translated - "Did not found indi_celestron_aux driver in a system. Also..

```
$ ls /usr/bin/indi_celestron_*
/usr/bin/indi_celestron_gps 
/usr/bin/indi_celestron_sct_focus
```

Version 2.0.3

rkaczorek, 2020-12-06:
Is the system updated from online repository?

luzik, 2020-12-06:
Was installed from image. After reporting here just installed via apt.