rkaczorek/astroberry-server #84: astroberry-server-wui (2.0.7)

AradoSKY, 2020-06-19:
Hello,

RPI3B+ and 2.02 astroberry are not working with nginx. Here is the evidence. When install this patch, nginix hangs up the process and does not finish. The /usr/share/indi/drivers.xml has a mismatched tag: Line 27, column 10. I fixed my verion. 
 
astroberry@astroberry:~ $ sudo dpkg --configure -a
Setting up astroberry-server-wui (2.0.7)
Synchronizing state of nginx.service with SysV service scrip with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable nginx
insserv: script klogd: service klogd already provided!
insserv: script klogd: service klogd already provided!

Ctrl-C cancels the joba as nginx never terminates and stays ready. This hangs up the patch process and does not close. 

dpkg: error processing pagage astroberry-server-wui (--configure)
installed astroberry-server-wui package post installation script subprocess was interrupted
Errors were encountered while processing:
  astroberry-server-wui

SYSLOG shows issues with KLOGD corresponding with this shutdown. The complaint is that overwriting a service already overwritten.

rkaczorek, 2020-06-19:
Upgrade to version 2.0.3 which has been released last week

AradoSKY, 2020-06-21:
Hello,

Thank you for your able assistance and reply. I was upgrading to 2.03 when this error occurred. I cannot upgrade without resolving “dpkg —configure -a”.  The 2.0.3 script hung up at this spot.

AradoSKY, 2020-06-22:
Hello,

I found a partial smoking gun. The wui Post Install step is the one which hangs. Specifically, "_systemctl restart indiwebmanager.service_" is the offending call. I am looking at what is causing this hang up. I am wondering if a AstroPI3 script issue has caused a problem. 

If I run "_/usr/bin/python3 /var/www/indiwebmanager/main.py -v -l /var/log/astroberry/indiwebmanager.log_" with terminal, I see no errors other than this stays resident in memory. Ctrl-C cancels the load. I puzzled why this works fine under RPI4 and RP3B+ crashes on the tarmac. The services are the same. Syntax is correct. This shows the activity. 

2020-06-21 18:40:22,544 - DEBUG: command line arguments: {'indi_port': 7624, 'port': 8624, 'host': '0.0.0.0', 'fifo': '/tmp/indiFIFO', 'conf': '/root/.indi', 'xmldir': '/usr/share/indi/', 'verbose': True, 'logfile': '/var/log/astroberry/indiwebmanager.log', 'server': 'standalone'}
2020-06-21 18:40:22,618 - INFO: pkill -9 indiserver
2020-06-21 18:40:22,646 - INFO: indiserver terminated successfully
2020-06-21 18:40:22,692 - INFO: using Bottle as standalone server
2020-06-21 18:40:22,793 - INFO: Deleting fifo /tmp/indiFIFO
2020-06-21 18:40:22,810 - INFO: indiserver -p 7624 -m 100 -v -f /tmp/indiFIFO > /tmp/indiserver.log 2>&1 &
2020-06-21 18:40:22,817 - INFO: echo "start indi_astrometry" > /tmp/indiFIFO
2020-06-21 18:40:22,825 - INFO: echo "start indi_gpsd" > /tmp/indiFIFO
2020-06-21 18:40:22,834 - INFO: echo "start indi_joystick" > /tmp/indiFIFO
2020-06-21 18:40:22,843 - INFO: echo "start indi_skysafari" > /tmp/indiFIFO
2020-06-21 18:40:22,854 - INFO: echo "start indi_qhy_ccd" > /tmp/indiFIFO
2020-06-21 18:40:22,864 - INFO: echo "start indi_asi_ccd" > /tmp/indiFIFO
2020-06-21 18:40:22,886 - INFO: echo "start indi_eqmod_telescope" > /tmp/indiFIFO
2020-06-21 20:01:00,941 - DEBUG: command line arguments: {'indi_port': 7624, 'port': 8624, 'host': '0.0.0.0', 'fifo': '/tmp/indiFIFO', 'conf': '/root/.indi', 'xmldir': '/usr/share/indi/', 'verbose': True, 'logfile': '/var/log/astroberry/indiwebmanager.log', 'server': 'standalone'}
2020-06-21 20:01:01,015 - INFO: pkill -9 indiserver
2020-06-21 20:01:01,044 - INFO: indiserver terminated successfully
2020-06-21 20:01:01,050 - INFO: using Bottle as standalone server
2020-06-21 20:01:01,134 - INFO: Deleting fifo /tmp/indiFIFO
2020-06-21 20:01:01,149 - INFO: indiserver -p 7624 -m 100 -v -f /tmp/indiFIFO > /tmp/indiserver.log 2>&1 &
2020-06-21 20:01:01,156 - INFO: echo "start indi_astrometry" > /tmp/indiFIFO
2020-06-21 20:01:01,164 - INFO: echo "start indi_gpsd" > /tmp/indiFIFO
2020-06-21 20:01:01,172 - INFO: echo "start indi_joystick" > /tmp/indiFIFO
2020-06-21 20:01:01,182 - INFO: echo "start indi_skysafari" > /tmp/indiFIFO
2020-06-21 20:01:01,202 - INFO: echo "start indi_qhy_ccd" > /tmp/indiFIFO
2020-06-21 20:01:01,240 - INFO: echo "start indi_asi_ccd" > /tmp/indiFIFO
2020-06-21 20:01:01,284 - INFO: echo "start indi_eqmod_telescope" > /tmp/indiFIFO
2020-06-21 20:01:31,848 - INFO: Exiting

From this, I can conclude that I have defined a profile in AstroBerry WEB and INDI is going to keep presenting it. This suggests that a Halt/Stop command is necessary after the restart command to prevent a profile from staying resident. I will alter WUI POST INIT to see if this resolves the issue of hanging up.

AradoSKY, 2020-06-22:
Hello,

Yes. Adding the SYSTEMCTL STOP command after the restart for a Defined AstroBerry WEB session resolved the DPKG --CONFIGURE -A issue. PI3B+ is back in business.

AradoSKY, 2020-06-22:
Hello,

I fixed the problem. If a indiwebmanager profile is created and set to auto run, auto run requests stay in memory. Auto run and update scripts don’t mix.  I added a systemctl stop indiwebmanager.service right the RESTART command for indiwebmanger. This resolved the hanging script.

closed 2020-06-22 by AradoSKY