rkaczorek/astroberry-server #85: Unable to "remote" solve

gaevoc, 2020-07-12:
Hello,
first of all my compliments for Astroberry, it is a nice initiative and deserves appreciation.

Being a novice, I hope you have a little time to help me with an issue related to _remote_ solving images. Astroberry is installed on a RPi 4 and I connect to INDI server from KStars running on a MacBook Pro. On the Astroberry INDI server I load the astrometry driver in the running profile.
Now, if I _Load and Slew_ an image using _remote_ solver option I always get **Solver failed. Try again** while if I take the very same image and launch _Load and Slew_ via VNC on the RPi, it works.
One more thing, the solver fails regardless of the current mount position.

Any suggestion?

Thank you
Gaetano

[image]

rkaczorek, 2020-07-12:
Hi Gaetano, thanks for the report.
This would be totally normal for _local_ astrometry.net plate solving, but... _remote_ seems really strange.
The only thing that comes to my mind is to check Options field in both scenarios. If there is no difference please report it to [INDI forum](http://indilib.org/forum.html). It could be a bug in KStars/Ekos.

gaevoc, 2020-07-13:
Hi Radek,
Thanks for your answer. 
Which are the Options on the Mac side that could have an impact on the remote solution?

G

rkaczorek, 2020-07-13:
I was wrong. I was having local plate-solving in mind. Remote plate solving with astrometry.net uses blind solving, so I cannot see any difference. Did you try to upload failing images directly to astrometry.net?

gaevoc, 2020-07-13:
I didn’t think of trying, if you think it is useful I will do it. Actually I feel it is not a problem related to the image because I am able to solve it locally on Astroberry server. 

G

wgauvin, 2020-07-26:
I had this issue and solved it.  It's to do with where it's getting the astrometry.cfg and also where it's trying to find the index files.

TL;DR: in `/etc/astrometry.cfg` add `add_path /home/astroberry/.local/share/kstars/astrometry` (or wherever you have installed your index files within Ekos on that Pi) as where to find the index files. Otherwise, you can add the additional setting your local Ekos of where to find you `astrometry.cfg` file on the remote server (That should be `/home/astroberry/.local/share/kstars/astrometry/astrometry.cfg`

When you VNC into the Astroberry and use the local astrometry-net Ekos adds the location of the config file when the command is executed. While a local plate solving is happening you can do a `ps -eaf|grep solve-field` and you will see that the there is a `--config /home/astroberry/.local/share/kstars/astrometry/astrometry.cfg` argument. This doesn't happen when you set up `remote` as the local Ekos doesn't know where the remote `astrometry.cfg` config is. You can test this by in your local client INDI Manager you can add all the arguments, including the `--config /home/astroberry/.local/share/kstars/astrometry/astrometry.cfg` (make sure you set all the other flags like high and low resolution, size, downsample, etc. Upload a file and submit (there is a threading issue that blocks the UI until the file is uploaded). On your Pi you will be able to see that the command is running. This should solve.

I found the simplest issue was to add a line in the default remote `astrometry.cfg` to say where the index files where.

I hope this helps.

gaevoc, 2020-07-26:
Hi Will,
this helps a lot!
Thank you so much for your suggestion, both the described methods work flawlessly. It is clear how the problem is behind the astrometry.cfg chosen to run the solving which is not the one containing the correct link to the index files repository. I choose to add the -config option to the solver custom options because I like the idea that I will use the very same config file regardless where I choose to solve, whether locally over VNC or remotely from the Mac.

wgauvin, 2020-07-26:
@gaevoc yeah it took me a full day to sort it out. in the Ekos documentation there was a online statement of that the index files had to be in the default location found in /etc/astronmetry.cfg

@rkaczorek perhaps we can add this to the wiki. Give people the options but it would save them hours/days scratching their head.

rkaczorek, 2020-08-05:
Added https://www.astroberry.io/docs/index.php?title=Astroberry_Server&action=submit#Plate_solving
@wgauvin please verify if it makes sense

wgauvin, 2020-08-06:
@rkaczorek  - the first part is clear, but the part of configuring the local Ekos is not clear (I know you're not trying to document how to use Ekos but in this case I think it would be helpful).

I would put something like this:
Otherwise, you can add the additional custom config setting in the local Ekos Solver Options of where to find your astrometry.cfg file on the remote server by using `--config <path-to-remote-astrometry.cfg>`. For a remote Astroberry server the config path should be /home/astroberry/.local/share/kstars/astrometry/astrometry.cfg

rkaczorek, 2020-08-06:
Corrected. Thanks for heads up on this

closed 2020-08-06 by rkaczorek