rkaczorek/astroberry-server #86: QHY Cameras not working after update

jamesbaillies, 2020-08-04:
I have updated my astroberry to the lastest verison and my qhy driver is not longer working. I revert back to an old image on astroberry and it worked. How to I recompile the QHY driver to work with the new update of raspberry pi and astroberry. Can you point me to the right place to download the files to make a new driver to work with the new pi kernel?

rkaczorek, 2020-08-05:
Please provide more info.
- What version of astroberry you use?
- What version of the driver works?
- What version of the driver does not work?

jamesbaillies, 2020-08-05:
Sorry Radek. I was very hasty with my first email. It works with Astroberry 2.0.2 with a fresh install of the image. I'm not sure which driver is installed in the image. If I do a sudo update and install on the fresh image the QHY drivers stop working. I've gone back to the Astroberry 2.0.2 image just so I can get some imaging done. There's nothing worse than a clear night and having to wrestle with techical issues! This situation drives me crazy and I'm thinking of never doing an update again. I supposed I should image this working version with all my settings just in case. Many thanks for your quick reply. James

rkaczorek, 2020-08-05:
That is fine. So the issue appears after upgrading 2.0.2 to 2.0.3? Is that right. I will give it a closer look over weekend.

jamesbaillies, 2020-08-05:
Yes that is correct. Thank Radek!

rkaczorek, 2020-08-09:
QHY drivers has been updated. Please upgrade your system and report back.

jamesbaillies, 2020-08-09:
Thanks Radek!

jamesbaillies, 2020-08-09:
Hi Radek,
I updated my system using sudo update and upgrade and it still does not
recognize the QHY 5ll camera I use as a guide camera. I did do a backup of
my previous setup first, so I can rollback to my previous install.
James

rkaczorek, 2020-09-07:
Check out this post https://indilib.org/forum/stellarmate/7641-new-guy-making-orion-ssag-pro-mono-function.html#59356